randdusing/cordova-plugin-bluetoothle #317: Cancelling a pending connect

RangerMauve, 2016-07-26:
I have a situation when sometimes devices take a *very* long time to connect to a peripheral. I would like to add some sort of logic for timing out the connect call so that a different device can connect to the peripheral instead (it might have a better signal). Currently there doesn't seem to be anything for doing that. I was wondering if calling `close` would cancel the connect, or alternatively if an API can be added for facilitating this use case.

randdusing, 2016-07-29:
Calling ```disconnect()``` or ```close()``` will cancel the connection. I usually recommend calling ```close()``` so you don't have to use ```reconnect()```.

RangerMauve, 2016-07-29:
Perfect. Thanks for the confirmation.

Kind of off topic, but should I also be closing when a device disconnects on its own?

randdusing, 2016-07-31:
Yep, closing tells your phone/tablet that you're no longer using the 'socket'

RangerMauve, 2016-07-31:
Cool. I think there was a case where I wasn't doing so when I had connection failures which might have led to some trouble

closed 2016-07-31 by RangerMauve