randdusing/cordova-plugin-bluetoothle #320: Service not correctly added

wenhsiu, 2016-08-02:
Hello, I'm completely new to this, so forgive me if this is a dumb question.
I'm writing an Android app that connects and sends data to raspberry pi 3.
As far as I understand, I'm supposed to add a new writable service to the pi, and writes to the characteristic. 
Correct me if I'm wrong, but these are the steps I've done:

Calls initializePeripheral after devices are connected -> discover -> addService -> discover -> write
The problem is, addService will return success, even though the discover afterwards discovered nothing new. 

One thing I noticed is, if I set the new service ID to be "1234", result.service will sometimes return "1234", but most of the time it returns "00001234-0000-1000-8000-00805f9-b34fb". Not sure why it's like this.

Is there something I did wrong? I feel like I must've misunderstood or missed out some parts of the process.

Thanks in advance!

randdusing, 2016-08-02:
I think you're mixing up central and peripheral functions. Check out the sections in the Readme titled Central Life Cycle and Peripheral Life Cycle. But it sounds like you want the Android app acting as a Central.

wenhsiu, 2016-08-03:
I see, I will look into it. Thanks so much!

closed 2016-08-03 by wenhsiu