randdusing/cordova-plugin-bluetoothle #321: Initialize never gives a result

Martijn-Jansen, 2016-08-03:
Hi Rand,
I'm trying to use the plugin to advertise a package. But it is just not working. The initialize of bluetoothle never gives a result. 
I have included a simple test project which is build in phonegap and then installed on my device (iphone 6  ios 9.3.3)
I hope you can help because i'm getting a bit desperate :)
[xsanyware.ble.zip](https://github.com/randdusing/cordova-plugin-bluetoothle/files/398973/xsanyware.ble.zip)

thxs
Martijn

randdusing, 2016-08-07:
You shouldn't be including bluetoothle.js. Cordova will automatically add in any plugin scripts. The code within ```onDeviceReady```looks good.

Martijn-Jansen, 2016-08-09:
thxs, that worked!

lepkifke, 2017-07-19:
I am having the same issue , first time i try to initialize it never gives result, second time it works...

randdusing, 2017-07-30:
@lepkifke How are you setting up the project? The Angular wrapper's example shows an example of how to set it up.

lepkifke, 2017-07-30:
The solution for me was to remove the restoreKey when working with IOS :   
Seems like something is not working when trying to set it to work from background thread...
var params = {
                    "request": true,
                    "statusReceiver": false
                    //, "restoreKey": "bluetoothleplugin"
                };