randdusing/cordova-plugin-bluetoothle #322: Subscribing error on iOS

timdelawe, 2016-08-04:
My App already works on Android, so now I try to run it on iOS (5s 9.3.3).

When I try to subscribe to a characteristic the error callback is invoked and gets the following object:
`{"service": ..., "name": ..., "characteristic": ..., "error": "subscription", "address": ..., "message": "Error Domain=CBATTErrorDomainCode=\"Unknown ATT error.\" UserInfo={NSLocalizedDescription=Unknown ATT error}"}` (... = correct value). 

The service and characteristic UUID's look like this: `XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX`. All of them are discovered correctly before I try to subscribe. I tried to use `discover()` and `services() + characteristics()`, but there is no difference.

randdusing, 2016-08-07:
I love iOS's error messages! Are you using notify or indication? It should automatically choose the right one, but I wonder if the characteristic is configured weirdly. What type of device is this? Can you print out the whole discover object that includes the properties and permissions?

timdelawe, 2016-08-08:
I use indication. The device is some special... It's used for communication with an industrial transceiver. That causes this individual characteristic. 

Here is a part of the discover object of the characteristics: `{"service": 2ABEC6C0-ADF7-11E3-A5E2-0800200C9A66, "name": Test, "status": "characteristics", "characteristics":[{"uuid": "F0189E80-ADF7-11E3-A5E2-0800200C9A66", "properties": {"indicate": true, "write": true, "read": true}},...]`.

The device has some more characteristics on this service. They all have the same properties and lead to the same issue. 

I tried to subscribe with and without the `isNotification` property, but nothing changed.

randdusing, 2016-08-09:
Yah isNotification was only for Android. Post the relevant code and I can review it. You can also try using the example app in the Angular wrapper https://github.com/randdusing/ng-cordova-bluetoothle.

timdelawe, 2016-08-09:
Here is the snippet: 
```
var params = {"address": address,
                    "service": SERVICE_UUID,
                    "characteristic": CHARACTERISTIC_UUID};

if (device.platform === "Android")
      params.isNotification = false;

bluetoothle.subscribe(onSubscribed, onError, params);
```
The error callback is invoked with the message I posted. 

This is executed before (works fine): 
```
bluetoothle.services(onServices, onError, 
         {"address": address, "services": []});

function onServices(objServices)
        {
            bluetoothle.characteristics(onCharacteristics, bleEvent.onError,
                                                {"address": address, "service": SERVICE_UUID,
                                                       "characteristics": []});        
        }

function onCharacteristics(objCharacteristics)
        {
            bluetoothle.descriptors(onDiscriptors, onError,
                                        {"address": address, "service": SERVICE_UUID, 
                                         "characteristic": objCharacteristics.characteristics[0].uuid});
        }

function onDiscriptors(objDescriptors)
        {
            subscribe();
        }
```

randdusing, 2016-08-12:
Is there more than one characteristic? If so, there could be a difference between CHARACTERISTIC_UUID and objCharacteristics.characteristics[0].uuid.

timdelawe, 2016-08-12:
There are multiple characteristics. 
I checked your guess: they are the same. I've also tried all other characteristics, without any success.

randdusing, 2016-08-13:
Besides the example app in the Angular wrapper, maybe try the LightBlue app?

timdelawe, 2016-08-17:
I tried your example app. It results in the same error message. 

I can't even debug through the error because the message is passed to the `callbackFromNative: function(callbackId, isSuccess, status, args, keepCallback)` function.

Can you post a link to the lightBlue app you mean?

randdusing, 2016-08-19:
LightBlue - https://itunes.apple.com/us/app/lightblue-explorer-bluetooth/id557428110?mt=8

timdelawe, 2016-08-24:
With the LightBlue app there is no error message but I don't receive any data.

randdusing, 2016-08-25:
Hmm, sounds like an issue between your device and iOS, not necessarily the plugin. Could you enable advanced debugging and see if you notice anything strange? https://developer.apple.com/bug-reporting/profiles-and-logs/

timdelawe, 2016-08-25:
I followed the instructions you've linked. Here is the part of the log file where the issue comes up: 
```
2016-08-25T07:57:36.520 XpcLeConnection.cpp:57    handleMsg            Notice     XPC        Received XPC message "CBMsgIdCharacteristicNotifyValue" from session "de.****.raddiag-central-202-0"
2016-08-25T07:57:36.521 GattGateway.cpp:461       setCharacteristicSubscription Notice     ATT        Subscribing to updates of characteristic handle 0x0010 on device "F9FABEF8-DA95-6454-3B24-5951A74A4D75"
2016-08-25T07:57:36.521 GattCommands.cpp:981      runCommand           Info       ATT        Found client configuration descriptor at handle 0x0012
2016-08-25T07:57:36.521 GattCommands.cpp:997      runCommand           Notice     ATT        Enabling indications on client configuration descriptor at handle 0x0010 on device "F9FABEF8-DA95-6454-3B24-5951A74A4D75"
OI_ERROR> ATT_GATT:1658  [65535]
2016-08-25T07:57:36.564 GattCommands.cpp:622      writeResponse        Error      ATT        Failed to write descriptor at handle 0x0012 on device "F9FABEF8-DA95-6454-3B24-5951A74A4D75" - result was BM3 STATUS 128
2016-08-25T07:57:36.565 GattCommands.cpp:64       notifyCommandCompleted Error      ATT        Command failed to execute with status BM3 STATUS 128
2016-08-25T07:57:36.565 GattCommands.cpp:64       notifyCommandCompleted Error      ATT        Command failed to execute with status BM3 STATUS 128
2016-08-25T07:57:36.565 XpcLeConnection.cpp:95    sendMsg              Notice     XPC        Sending XPC message "CBMsgIdCharacteristicValueNotifying" to session "de.****.raddiag-central-202-0"
2016-08-25T07:57:37.603 GattCommands.cpp:743      runCommand           Notice     ATT        Reading GAP name characteristic value on device "F9FABEF8-DA95-6454-3B24-5951A74A4D75"
2016-08-25T07:57:37.675 GattDatabase.cpp:567      addCharacteristicValue Info       ATT        Found service "0x0054 Primary Service      [ UUID: 0x1800, endHandle: 0x0058 ]" for value "0x0056 Characteristic Value [ 49 46 53 2D 42 54 2$
2016-08-25T07:57:37.675 GattCommands.cpp:651      updateName           Notice     ATT        The device "F9FABEF8-DA95-6454-3B24-5951A74A4D75" is named „****“
2016-08-25T07:57:37.676 LeDeviceCache.mm:530      -[LeDeviceCache writeDevice:] Info       LE         Writing device "F9FABEF8-DA95-6454-3B24-5951A74A4D75" to 'other devices' cache
2016-08-25T07:57:37.720 LeDeviceCache.mm:585      -[LeDeviceCache writeDevice:] Info       LE         Successully wrote device "F9FABEF8-DA95-6454-3B24-5951A74A4D75" into 'other devices' cache
```

Around this part everything looks fine.

randdusing, 2016-08-25:
How easy is it to modify the code on the BLE device? I wonder if you could switch over to notification and see if the issue persists.  The iOS [subscribe code](https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/src/ios/BluetoothLePlugin.m#L1359) is pretty simple at the plugin level, so I'm not sure if there's much I can change.

timdelawe, 2016-08-25:
Sorry the device is protected and developed by somebody else than me. But I got another hint today. So I think this has nothing to do with the implementation of your plugin. 

The BLE device seems to be more "special" then I thought: It only allows reading characteristics through notifications but not directly. Have you any idea for a workaround?

randdusing, 2016-08-26:
I'm not really sure what you mean by your last sentence. Isn't that what you're trying to do? Read via notification and/or indication?