randdusing/cordova-plugin-bluetoothle #323: connect() do not have any callback

wangxiao, 2016-08-07:
I called `bluetoothle.connect`, but it do not have any callback, is connecting need a very long time (> 1min) ？

My code like this:
```
var connectTimes = 0;
var scan = function() {
  bluetoothle.startScan(function(res) {
    if (res.address && !connectTimes) {
      connect(res.address);
      stopScan();
      connectTimes ++;
    }
  });
};

function stopScan() {
  bluetoothle.stopScan();
}

function connect(address) {
    bluetoothle.connect(function(res) {
        alert('ok');
    }, function(err) {
       alert('T-T');
    }, {
      address: address,
    });
}
```

randdusing, 2016-08-07:
Hmm, the code looks fine. You're able to see it in the scan results, but trying to connect just times out?

wangxiao, 2016-08-07:
Thx, the results also look good.
I think i need check the native android java code with my android developer friends.

randdusing, 2016-08-07:
Try using something like LightBlue to see if you can connect with other apps on the same phone.

wangxiao, 2016-08-07:
Yes, it is ok. Use the official native android App of the bluetooth device can connect.

wangxiao, 2016-08-07:
After long time, maybe 10 min, show the connect failed.

closed 2016-08-07 by wangxiao

wangxiao, 2016-08-07:
android version 5.1, connect always failed.   https://github.com/randdusing/cordova-plugin-bluetoothle/issues/185

hazemhagrass, 2016-10-29:
Any updates here!

randdusing, 2016-10-31:
@hazemhagrass I need more details!

hazemhagrass, 2016-10-31:
Here is a sample of my code:
`$cordovaBluetoothLE.initialize({request: true}).then(null, function (obj) {
                console.log("INITIATE BLUETOOTHLE ERROR : " + JSON.stringify(obj));
            }, function (obj) {
                console.log("INITIATE BLUETOOTHLE SUCCESS : " + JSON.stringify(obj));

                function onBondCallback(obj) {
                    if (obj.status !== 'bonded') {
                        return;
                    }
                    console.log("BONDING BLUETOOTHLE : " + JSON.stringify(obj));
                    isLocationEnabled(function () {
                        console.log("START BLUETOOTHLE CONNECTION TO DEVICE: " + $scope.hrm.address);
                        $cordovaBluetoothLE.connect({
                            address: $scope.hrm.address,
                            autoConnect: true
                        }).then(null, function (obj) {
                            console.log("BLUETOOTHLE CONNECTION ERROR : " + JSON.stringify(obj));
                        }, function (obj) {
                            console.log("BLUETOOTHLE CONNECTION SUCCESS : " + JSON.stringify(obj));
                            if (obj.status == 'connected') {
                            } else if (obj.status == 'disconnecting' || obj.status == 'disconnected') {
                            }
                        });
                    });
                }

                function onUnbondingCallback(obj) {
                    console.log("UNBONDING BLUETOOTHLE : " + JSON.stringify(obj));
                    $cordovaBluetoothLE.bond({address: $scope.hrm.address}).then(null, onBondCallback, onBondCallback);
                }

                if (ionic.Platform.isAndroid()) {
                    $cordovaBluetoothLE.unbond({address: $scope.hrm.address}).then(onUnbondingCallback, onUnbondingCallback);
                }
                else {
                    onBondCallback({status: 'bonded'});
                }

            });`

hazemhagrass, 2016-10-31:
@wangxiao did you solve your problem? seems that I have an issue just like you

randdusing, 2016-10-31:
@hazemhagrass Details besides the code as well, what device, version, etc?

wangxiao, 2016-11-01:
@hazemhagrass  sorry, i have not solve the problem. Now i have developed Android native app, and found a Android engineer.