randdusing/cordova-plugin-bluetoothle #324: sending more than 20 byte

mjseyyedi, 2016-08-08:
I tried writeQ function to send more than 20 bytes but it didn't work for me. of course I think i did sth wrong here is my send function:

```
$scope.sendText = function (inputText) {
        var bytes = bluetoothle.stringToBytes(inputText);
        var encodedString = bluetoothle.bytesToEncodedString(bytes);
        if (inputText.length < 20) {
          bluetoothle.write(
            function (result) {
              alert("data transmitted" + JSON.stringify(result));
            },
            function (error) {
              alert("data transmission failed" + JSON.stringify(error));
            },
            {
              value: encodedString,
              service: $scope.sendCharService.serviceId,
              characteristic: $scope.sendCharService.uuid,
              type: "noResponse",
              address: $scope.deviceId
            }
          );
        }
        else {
          bluetoothle.writeQ(
            function (result) {
              alert("data transmitted" + JSON.stringify(result));
            },
            function (error) {
              alert("data transmission failed" + JSON.stringify(error));
            },
            {
              value: encodedString,
              service: $scope.sendCharService.serviceId,
              characteristic: $scope.sendCharService.uuid,
              type: "noResponse",
              address: $scope.deviceId
            }
          );
        }
      }
```

what am I missing?

randdusing, 2016-08-09:
Looks fine to me, are you receiving any callbacks?

mjseyyedi, 2016-08-13:
@randdusing yes , "data transmitted" shows up. means worked perfectly

randdusing, 2016-08-14:
Not sure if I follow. It's working for write but not writeQ. The writeQ callbacks are never called?

mjseyyedi, 2016-08-14:
the writeQ callback calls correctly , but in bluetooth I don't receive more than 20 bytes

randdusing, 2016-08-19:
On the peripheral side, only 20 bytes show up?