randdusing/cordova-plugin-bluetoothle #326: Start advertising with manufacturer data

Martijn-Jansen, 2016-08-09:
Hi Rand,
First of all, you're plugin works great! thxs for that.
My question has to do with advertising data in peripheral mode. i've got an example of advertising data with xamarin.forms. I need a cordova implementation of this example because the hardware we are using only implements this example.

```
public void Start(string payload)
		{
			var manager = (BluetoothManager)Android.App.Application.Context.GetSystemService(Context.BluetoothService);
			var adapter = manager.Adapter;
			if (adapter != null)
			{
				if (adapter.State == State.On)
				{
					_publisher = adapter.BluetoothLeAdvertiser;

					var settings = new AdvertiseSettings.Builder();
					settings.SetAdvertiseMode(AdvertiseMode.Balanced);

					var data = new AdvertiseData.Builder();
					**data.AddManufacturerData(0x004C, HexStringToByteArray(payload));**

					_publisher.StartAdvertising(settings.Build(), data.Build(), _callback);
				}
			}
		}
```
My question is about the AddManufacturerData. 
Is it possible to do something like this with you're startAdvertise implementation.

Thxs!

Grtz,
Martijn

randdusing, 2016-08-12:
Yep, you can. I thought I had it documented, but it's not. Here's the relevant Java code:

```java
int manufacturerId = obj.optInt("manufacturerId", 0);
byte[] manufacturerSpecificData = getPropertyBytes(obj, "manufacturerSpecificData");
if (manufacturerId >= 0 && manufacturerSpecificData != null) {
  dataBuilder.addManufacturerData(manufacturerId, manufacturerSpecificData);
}
```

Modify params to include these properties: manufacturerId is just an integer. manufacturerSpecificData should be base64 encoded array of bytes. I may be doing plugin work this weekend and can possibly produce an example.

randdusing, 2017-07-16:
Documentation hasn't been updated, but you just need to add an ID for manufacturerId and a data string for manufacturerSpecificData, similar to write date.

anilkunchalaece, 2017-07-17:
how to understand the manufacturing data ?
```
      var paramsAdvertise = {
      //services: ["1234"],
      //service: "1234",
      //name: "nil_k",
      //mode: "lowLatency",
      connectable: false,
      timeout: 500,
      manufacturerId: 0x0201,
      manufacturerSpecificData:0x0a
    };
```

output is : 0x0102D7

and i don't understand why D7 is coming ?

randdusing, 2017-07-17:
manufactuerSpecificData accepted a base64 encoded string of bytes

omairsaleem, 2017-07-26:
Hi Martin,

Thanks for the Xamarin Forms example above. Could you please tell me how you are declaring the _callback parameter for StartAdvertising method above? Thanks.