randdusing/cordova-plugin-bluetoothle #327: Android startAdvertising error

Gianni2013, 2016-08-11:
Hello Rand,
I have a problem with the BLE advertising on Android, when I call startAdvertising, the error message is "advertising is not supported".
It 'a problem of my smartphone, Android or implementation of the plugin?

I call:
1) bluetoothle.initialize(resolve, { request: true, statusReceiver: false });
2) bluetoothle.initializePeripheral( 
			function( res ) { 
				alert( 'initialize ok ' + JSON.stringify( res ) );
				advOn();
			},
			function( res ) { alert( 'initialize err ' + JSON.stringify( res ) ) },
			params);

and on success of initializePeripheral I call
	bluetoothle.startAdvertising(
		function( res ) { alert( "adv success " + JSON.stringify( res ) ) }, 
		function( res ) { alert( "adv error " + JSON.stringify( res ) ) }, 
		paramsService);
	
Thanks!
Gianni

randdusing, 2016-08-12:
What phone and Android version?

Gianni2013, 2016-08-12:
Android 5.0.1
Huawei P8

randdusing, 2016-08-14:
Are you developing locally? I think I found the problem in startAdvertisingAction(). Can you remove the ```|| !bluetoothAdapter.isMultipleAdvertisementSupported()``` and try that? I'm not sure why I added this check. It would really only make sense if there was another API call to determine whether the chip is already advertising. All three of my Android devices must support multiple advertisement since I haven't ran into this problem.

```
    BluetoothLeAdvertiser advertiser = bluetoothAdapter.getBluetoothLeAdvertiser();
    if (advertiser == null || !bluetoothAdapter.isMultipleAdvertisementSupported()) {
      JSONObject returnObj = new JSONObject();

      addProperty(returnObj, "error", "startAdvertising");
      addProperty(returnObj, "message", "Advertising isn't supported");

      callbackContext.error(returnObj);
      return;
    }
```

Gianni2013, 2016-08-19:
Hi Rand,
I tried the code, without modification, on samsung and it works perfectly. So I understand (but not read anywhere) does not work on HUAWEI.

Thank you!

randdusing, 2016-08-19:
Cool glad it works. I'm going to keep this open to remind myself to remove that check for multiple advertisement support.

mjseyyedi, 2016-08-24:
what is the service value in startAdvertising?? whatever I passed, was failed!

randdusing, 2016-08-25:
It's one of the services that you added via addService.

petearvanitis, 2017-03-18:
I ran into the same issue w/ my NVIDIA Sheild tablet. I verified that the code works via my S7 Edge (which supports advertising).  I tracked down this list of android devices that shows which ones support advertising mode in case someone else runs into this issue. https://altbeacon.github.io/android-beacon-library/beacon-transmitter-devices.html

commit 2ffef2ca (2020-04-10): Fixes #327
Finally, as the author suggest, by removing the condition "|| !bluetoothAdapter.isMultipleAdvertisementSupported()" 2 of my old android cellphones start working, because they support advertising but the condition isMultipleAdvertisementSupported returned false for both of them.
commit 8232114a (2020-04-10): Merge pull request #587 from aparraga/#327

Fixes #327
closed 2020-04-10 by randdusing