randdusing/cordova-plugin-bluetoothle #329: Android Discover - old services?

dibari, 2016-08-16:
I have a device that is set in either normal functioning mode or recovery mode for firmware updating.  Each mode has a unique service and characteristic set.  When connecting to a device in recovery mode, it almost always populates the services and characteristics (through discover) as to what they are in the normal mode.

Is it possible discover() is caching or saving the services and characteristics?

randdusing, 2016-08-19:
Are you seeing this on both platforms? iOS does have a didModifyServices event, but it's not implemented in the plugin. Are you closing the connecting in between restarting the device? Closing the connecting should force it to actually rediscover. I did something very similar, but never ran into this problem. I'm assuming because I always fully closed out the connection.

dibari, 2016-08-22:
I'm only seeing the issue on android, and I'm closing out the connection completely.  I am even uninstalling and reinstalling the app to be sure.  However, the device still has the old services.  I believe it may be a version issue.  It doesn't work properly on API 19, but does on 23.

Is this stackoverflow question possibly related to an android caching issue : http://stackoverflow.com/questions/22596951/how-to-programmatically-force-bluetooth-low-energy-service-discovery-on-android

randdusing, 2016-08-22:
Stackoverflow link should help. It looks like I can just call the refresh method through reflection each time connect/reconnect is called.

PRs are welcome, otherwise I probably won't get to this until September at the earliest.

dibari, 2016-09-01:
I implemented what seems to be a working model, so I opened a pull request.  I'm only able to test it on two devices, but it seems to be working.  You'll see I added in flag on the discover service.

randdusing, 2016-09-10:
Merged! Let me know if you see any issues.

closed 2016-09-10 by randdusing