randdusing/cordova-plugin-bluetoothle #331: Trying to write single byte value however the BLE peripheral doesn't seem to take it

chiefie, 2016-08-19:
Hello,

The BLE peripheral that I'm trying to write to (a specific characteristic - C1 ) is expecting a byte value (e.g. 0x00 or 0x01 etc).

I've tried following the following examples:

- https://github.com/randdusing/cordova-plugin-bluetoothle/issues/287#issuecomment-221470087
- https://github.com/randdusing/cordova-plugin-bluetoothle/issues/67#issuecomment-53074357
- https://github.com/randdusing/cordova-plugin-bluetoothle/issues/100#issuecomment-61998549

There is another characteristic (C2) (on the same BLE peripheral) will be updated in respond to the write event of the previous characteristic (C1) to confirm the write request has been made and updated.

However, upon reading C2, it is still showing the previous value.

YET, when using LightBlue Explorer to write to C1, and later reading C2, the value on C2 is properly updated. The write format is in Hex.

I'm at puzzled and would appreciate if someone could give an insight how the write method actually work?

p/s: I'm using ng-cordova-bluetoothle with Ionic Framework v1.3.x on iOS.

randdusing, 2016-08-19:
Possibly something is wrong with your write call. Can you post your write related code?

chiefie, 2016-08-20:
This is the write method.

`_requestNodeIdx = function (deviceAddress, moduleIdx) {
      return function () {
        console.log('Request Node Idx: ' + moduleIdx);

        var u8 = new Uint8Array(1);
        u8[0] = moduleIdx;
        var encodedString = $cordovaBluetoothLE.bytesToEncodedString(u8);

        var params = {
          address: deviceAddress,
          service: BLE.LDS_NODE_INFO_UUID,
          characteristic: BLE.LDS_NODE_INFO_CHARACTERISTICS.NODE_INDEX_REQUEST.UUID,
          value: encodedString,
          type: 'noResponse',
          timeout: BLE.TIMEOUT
        };

        console.log('Request Node Idx params: ' + JSON.stringify(params));

        // return $cordovaBluetoothLE.write(params);

        var q = $q.defer();

        $cordovaBluetoothLE.write(params).then(
          function (obj) {
            console.log('write [SUCC]');
            console.log(JSON.stringify(obj));

            q.resolve(obj);
          },
          function (obj) {
            console.log('write [FAIL]');
            console.log(JSON.stringify(obj));

            q.reject(obj);
          }
        );

        return q.promise;
      };
    };`

chiefie, 2016-08-24:
Hello @randdusing is there any update about this? Still hoping if this is possible or did I do it correctly when sending Byte (hex value).

randdusing, 2016-08-25:
Hey, must have deleted the email notification. The write function seems fine assuming moduleIdx is either 0x00 or 0x01. Are you maybe reading too early before the write has time to 'process'?

chiefie, 2016-08-25:
Hello.

The value for moduleIdx is integer (e.g. 0 or 1). Should it be formatted to 0x00 or 0x01, as string or straight hexidecimal number? *hmmm that could be it*

I have added a sleep method between write and read and it was not accepted by the BLE peripheral.

I will give it another try with supplying the value as you mentioned (0x00).

Cheers.

randdusing, 2016-08-25:
Hexadecimal or decimal should work fine. What do you mean by a sleep method was not accepted by the BLE peripheral? Maybe post your combined write and read code?

chiefie, 2016-08-25:
I tried integer (decimal) and it still doesn't get written.

The sleep method I had in place between write and read is just a $timeout. I had it set to as high as 10s. However, this is not the issue here. I checked the BLE peripheral using LightBlue Explorer app (iOS) on the characteristic that I read from to confirm the update, it did not receive the update I sent.

However, if I use LightBlue Explorer app to write then later to read in hexidecimal, it gets written successfully.

randdusing, 2016-08-25:
Oh, I just noticed you are writing with noResponse. Can you try testing without that? The characteristic may not support writeWithoutResponse. You can also print out the discovery object and check the characteristics properties. Or LightBlue should tell you if you're writing or writing without response.

chiefie, 2016-08-25:
Oh @randdusing removing that property from the params worked!! WOW!! Thank you very much.

It's interesting when I first coding it, I tried using without it and it didn't work but now it does. Either way, thank you very much. It is quite a learning curve but I'm satisfied with the process and learning.

Worked brilliantly.

closed 2016-08-25 by chiefie

randdusing, 2016-08-26:
Cool glad it's working