randdusing/cordova-plugin-bluetoothle #334: steps of receiving data

mjseyyedi, 2016-08-24:
I'm totally new to BLE , could you please give me steps of receiving data . I mean what functions should I call and in what order to get the value of sent data from bluetooth device?

randdusing, 2016-08-25:
The [readme](https://github.com/randdusing/cordova-plugin-bluetoothle#central-life-cycle) shows the high level steps for connecting and reading to the device. Also, check out the ionic based example in the [angular wrapper](https://github.com/randdusing/ng-cordova-bluetoothle).

closed 2016-08-25 by randdusing

mjseyyedi, 2016-08-27:
I've read all these docs before, but I explain the problem to you , maybe you could help me.
I can connect and write successfully to my ble board. but when I send data from board to my phone , no event in initializePripheral Callback triggers! but when I use nordicsemi's official android app , I receive it. I don't know what I'm missing. should I subscribe to any service for receiving data or what??

randdusing, 2016-08-28:
initializePeripheral is only if your app is acting as a peripheral. Is you app supposed to be a peripheral or central?

mjseyyedi, 2016-08-29:
I think it's central. What other way is for central app

mjseyyedi, 2016-08-29:
I mean how can I read data, in my central device?

randdusing, 2016-08-30:
I can't really provide one on one help unfortunately. Please checkout the readme and the example. If you have any specific questions, please post the relevant code.