randdusing/cordova-plugin-bluetoothle #335: createBond not returning when device bound to other phone

maximilianloy, 2016-08-29:
createBond is working unless the app tries to bond with a device which has already been bonded to a different phone. isBonded will return false in that case (as the phone has no key stored yet) but createBond never returns. It is stuck, probably waiting for the device to respond.

Would be great to have a timeout parameter also on the createBond operation for such a case.

This was tested on a Nexus 9 and Nexus 5 both running Android 6.0.1

randdusing, 2016-08-30:
It's been a while since I've looked at bonding. I would have thought it would just fail if the device denied the bond. I'll look into adding a timeout, but currently with all the other actions, the timeout is done on the Javascript/Angular side.

maximilianloy, 2016-08-30:
It might be that other devices do that, I don't know. Maybe s.o. could verify that on a different type of device.