randdusing/cordova-plugin-bluetoothle #337: UUID upper/lowercase inconsistencies between platforms

maximilianloy, 2016-08-30:
Switching to the latest version (from 3.0.0) I ran into inconsistencies regarding the UUID upper/lowercase naming between Android and iOS. I am using the same subscription callback for all characteristics and switch / case on obj.characteristic, but iOS returns uppercase UUIDs, whereas Android returns lowercase UUIDs!

Minor issue as it can easily be circumvented using toUpperCase/LowerCase().

randdusing, 2016-09-03:
Should be an easy fix, but it could potentially be breaking to people who may only be targeting Android and expecting lowercase. Not sure if I should increment the major version or keep it minor.

randdusing, 2016-09-10:
Added this to the latest master! It will be part of the next release.

closed 2016-09-10 by randdusing