randdusing/cordova-plugin-bluetoothle #339: Bluetooth not initialized error on android

Rohithastak, 2016-09-01:
Hi Radd,
This plugin working great in IOS but in my all android device its throwing error of Bluetooth not initialized even though my bluetooth is active. I've gone through the documentation but still facing same problem. please help.

randdusing, 2016-09-02:
Can you post your initialize related code?

Rohithastak, 2016-09-02:
var paramsObj = {address: localStorage['address_key']}; // localStorage['address_key'] this is UUID of device
bluetoothle.reconnect(objHR.reconnectSuccess, objHR.reconnectError, paramsObj);

 reconnectSuccess: function(obj)
            {
                if (obj.status == "connected")
                {
                    //alert("Reconnected to : " + obj.name + " - " + obj.address);
                    //alert("rECONNECT Beginning discovery");
                    //bluetoothle.discover(objHR.discoverSuccess, objHR.discoverError);
                    var paramsObj;
                    var strDeviceType = sessionStorage.platform;
                    switch (strDeviceType)
                    {
                        case 'iOS':
                            //alert("IPHONE HERE : "+obj.address);
                            //var arrServices = new Array();
                            //arrServices.push(strServiceUuid);

                            paramsObj = {address: localStorage['address_key'], serviceUuids: []};
                            bluetoothle.services(objHR.servicesSuccess, objHR.servicesError, paramsObj);

                            break;

                        default:
                            //alert("ANDROID HERE");
                            paramsObj = {address: localStorage['address_key']};
                            bluetoothle.discover(objHR.discoverSuccess, objHR.discoverError, paramsObj);
                            break;
                    }

                } else if (obj.status == "connecting")
                {
                    //alert("Reconnecting to : " + obj.name + " - " + localStorage['address_key']);
                }
                else
                {
                    alert("Unexpected reconnect status: " + obj.status);
                    objHR.disconnectDevice();
                }
            },
            reconnectError: function(obj)
            {
                alert("Reconnect error: " + obj.error + " - " + obj.message);
                objHR.disconnectDevice();

            },

Its is giving me "Reconnect error : Bluetooth not initialized"..

randdusing, 2016-09-03:
You need to make a call to bluetoothle.initialize() before trying to connect.
See [here](https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/readme.md#initialize)

Rohithastak, 2016-09-08:
Hi Randd,
I have given permission to device using  bluetoothle.requestPermission(requestPermissionSuccess, requestPermissionError)
Now its working fine,
Thanks

randdusing, 2016-09-10:
Cool, glad it's working

closed 2016-09-10 by randdusing