randdusing/cordova-plugin-bluetoothle #342: Android 5.1: Discover method returning exception.

DaniloCouto, 2016-09-05:
After a succeeded connection with an Bluetoothle device, I tried to discover his services and characteristics.
But, sometimes , the method returns an Java Exception : 
`Attempt to invoke virtual method 'java.lang.String java.lang.Object.toString()' on a null object reference`

It seems that the wasNeverConnected Native Java method is returning Null without throwing an callback. and during the discoverAction Method  it's impossible to convert a null object to string

randdusing, 2016-09-05:
Do you know what line it's throwing the error?

DaniloCouto, 2016-09-06:
On 1615 line of BluetoothLePlugin.java
I've even put a log to see if the `connection.get(keyDiscoveredState)` was really null on .toString.
`Log.v("MyBlueTest", "connection = " + connection.get(keyDiscoveredState));`
And yes, the log returned null.

But this error is odd and happen hardly ever.
And after that error if I try to restart the bluetooth by turning on and off, the bluetooth just... dies, and I need to restart the phone.

hannibaldinski, 2016-09-08:
Not sure but this might be related.

When I try connecting and it fails. The second time I try to connect and it succeeds, the discover function will return NULL in its error callback. I can fix it by calling close() and then connect() again.

[image]

randdusing, 2016-09-10:
You shouldn't be able to call connect back to back. Are you calling close in between? Or are some of those actually reconnect calls? If that's reconnect, I think I know the issue and will fix it this evening!

randdusing, 2016-09-10:
Actually just committed the change with master, I'm curious if you can still recreate.

randdusing, 2016-09-11:
Please re-open if that doesn't fix it

closed 2016-09-11 by randdusing