randdusing/cordova-plugin-bluetoothle #343: Initialize callback is only called once and not on every BT status change.

corneyl, 2016-09-06:
When I call the initialize function, the callback is called as expected, but when I disable or enable Bluetooth, it's not called again. I tried enabling BT using the device settings and using the `enable()` function, disabling only using the device settings.

I'm using the following code:
```javascript
var that = this;
// Init bluetooth, and define the status receiver function
bluetoothle.initialize(function(result) {
    if (result.status === "enabled") {
        console.log("Bluetooth is enabled.");
        
        // Always call handleScan() to start scanning.
        that.handleScan();
    }
    else {
        // Bluetooth is disabled, try to enable
        console.log("Bluetooth is disabled.");
    }
    
    // Broadcast the enabled and initialised event
    if(!that.initialised) {
        that.initialised = true;
        that.ev.emit('initialised');
    }
    that.enabled = (result.status == 'enabled');
    that.ev.emit('enabled', result.status);
    
}, this.handleError, {
    request: false,
    statusReceiver: true,
    restoreKey : "abc123"
});
```
And the following device:
- Samsung Note 10.1 (SM-P600)
- Android 5.1.1

randdusing, 2016-09-10:
I no longer have a 5.* device, but this is working fine on my Nexus 5X and Samsung Tab A running 6.*. Can you try using the example app that comes with the Angular wrapper? That's the quickest way to rule out custom code issues.

sunny-mittal, 2017-01-18:
I'm having this issue as well on my Galaxy S7 running Android 6.0.1. Oddly, it used to work just fine and I haven't modified the plugin or my initialize code at all, so I'm guessing it's a state that's possible to get into but whose cause is unclear. If anyone encounters and resolves this, please post here :)

randdusing, 2017-01-19:
Are you always calling initialize multiple times? That may be leading to an inconsistent state.

sunny-mittal, 2017-01-19:
I wrote a wrapper for the `initialize` call and I have yet to figure out why the callback doesn't fire on occasion but, of course, when trying to repro and debug the issue, it stopped being an issue, so for the time being, I'm going to call it good :) In any case, I can confirm it's not an issue with the plugin; I'm able to test in an isolated environment and observe proper functionality on 7 devices. I only posted here to get feedback from others in case they observed it as well.