randdusing/cordova-plugin-bluetoothle #344: Subscribe Success Callback  Method call only once

Rohithastak, 2016-09-08:
Hello ,
when i am calling bluetoothle.subscribe(objHR.subscribeSuccess, objHR.subscribeError, paramsObj)
this only gives me result.status == "subscribed" and do nothing after that, it doesnot give result "subscribedResult". Subscribe callback method is called only once.
 please share what to do next.

randdusing, 2016-09-10:
What type of BLE device, phone/tablet, Android version, etc? There's a PR for changing the subscription code that I haven't fully evaluated, but you may want to check it out to see if it works.

Rohithastak, 2016-09-12:
BLE device : Wahoo's Heart Rate Monitor(TICKER B3E4)
Phone : Samsung galaxy g7
Android version : 6.0

randdusing, 2016-09-13:
Try out that PR. It works for the requestor, and I'm hoping to review it within the next week or so. If it works for you, that will give me more confidence!

phice, 2016-11-03:
same for me, by setting "clearCache": false I can get the callback but I cannot subscribe anymore apparently

rgstephens, 2017-10-28:
I'm having the same issue.