randdusing/cordova-plugin-bluetoothle #345: OSX and Angular

knoandco, 2016-09-12:
Dear Rand,
Your plugin is really amazing. Thanks a lot for sharing such a great work.
However, I have a question. I am trying to compile an app on OSX. That app works perfectly on Android and iOS and I thought it was time to try on OSX and push a bit some limits.
Unfortunately, I can't use your plugin with Angular on OSX. window.bluetoothle is not "supported". Would you have a piece of advice to make it work ? 
Thanks for your time and sharing.
David

randdusing, 2016-09-13:
Are you using https://github.com/randdusing/ng-cordova-bluetoothle? I just tried it out with the latest plugin and angular wrapper version, and it's working fine. But I did run into issues with it back in March. The plugin would frequently just fail to run. I figured it was something Cordova related since other plugins weren't loading either.

Long story short, try out the Ionic based example in the Angular wrapper and see if that works for you.

knoandco, 2016-09-13:
Thanks for the tips.
Again, wonderful work and I'll my best to try that by Thursday.
Cheers,
D.

knoandco, 2016-09-14:
Dear Rand,

Thanks! I have tried the example provided and it worked perfectly.

 So now, just need to understand the difference with my app. Again, amazing work!

David

closed 2016-09-14 by knoandco