randdusing/cordova-plugin-bluetoothle #349: Ionic Native Plugin Wrapper

ikarus-balsero, 2016-09-19:
Hello, first I would like to give you kudos for this great plugin. Ionic 2 is almost in production and I was wondering if you are considering creating a wrapper to be use with Ionic Native? Thanks in advance!!

randdusing, 2016-09-20:
I haven't used Angular2 or Ionic2 very much so far. I have an existing Angular wrapper that you could probably modify to support Angular2. I think you need to replace the promises with notify callbacks to observables.

maximilianloy, 2016-10-25:
I would like to +1 this.

Ionic2 RC is out now and I am considering migrating my BLE app as I saw great performance improvements with another app.

randdusing, 2016-10-26:
Any recommendations on examples or tutorials to use? The actual wrapper is probably pretty easy to implement if you know Angular 1 and Angular 2, but the example app (that also helps test it) would be a little more difficult for Angular 2 noob like me.

ikarus-balsero, 2016-10-27:
Hi Rand,

This tutorial is pretty straight forward, highly appreciated if you can dedicate some time to do this.
https://github.com/driftyco/ionic-native/blob/master/DEVELOPER.md

Like @maximilianloy said, Ionic2 RC is out and there are substancial performance improvements. 

The only reason I haven't migrated my apps to it is due the BLE functionality, the BLE plugin supported by Ionic Native is not as good as yours! 

Thanks in advance!

randdusing, 2017-01-22:
I don't do much Ionic / BLE work anymore, so I don't have any plans to create an Angular 2 wrapper. The wrapper isn't really complex, so it should be easy to convert from Angular 1 to 2. Unfortunately, I don't have any experience with Angular 2 and not sure when I'll have the time to pick it up.