randdusing/cordova-plugin-bluetoothle #351: Support for LE Secured mode feature in BLE 4.2

jpatel88, 2016-09-22:
I have a custom BLE peripheral device implementing LE Secured mode from BLE 4.2 spec. The only way I can get it to connect is if I make the peripheral device initiate the authentication/pairing request and this works for iOS and android. Downside is the pairing request dialog that android throws up and user has to tap on "agree" to proceed....

Is there any plan to add support for initiating the pairing request from the mobile app/plugin so that the pairing request for LE Secured mode can be automated from the mobile app without having the user to tap on anything else??

BTW - I like this plugin!!

randdusing, 2016-09-23:
On Android, you may be able to use ```bond()``` but I'm pretty sure it still prompts the user to confirm. I don't see anything new in the API for Android 7.0  that would change this. Same with iOS 10, not seeing any changes.

NunoSav, 2019-06-03:
I'm having an issue related to this one.

Some characteristics i need to read from a device are encrypted and while the `bond()` method works great on Android with iOS i don't seem to have anything of that sort.

@randdusing Is there still no way to either request pairing on iOS or get a hold of the iOS pairing alert response by the user? Do i have to guess when and what the user chooses? Thanks for your time.

randdusing, 2019-06-07:
Still nothing that I know of.