randdusing/cordova-plugin-bluetoothle #352: about read() and respond()

zhuqingsong59, 2016-09-23:
i am know little about bluetooth,if i want to get data from bluetoothle device ,how many times should i call function read(),just one or use timer to call many times? and about respond(),you say it can use to Respond to a read or write request,this request mean the request of my app?

randdusing, 2016-09-23:
You can call ```read()``` as often as you want, but usually read attributes don't change too often. You're probably better off using ```subscribe()```. You don't need to use ```respond()``` unless you're trying to turn your phone into an advertising peripheral.

zhuqingsong59, 2016-09-26:
thanks a lot ,but when i call read() or subscribe(),there is more than one serviceUUID and characteristicUUID , i must ergodic each UUID,then get data that i need by value?

randdusing, 2016-09-27:
Not sure if I totally understand your question, but you can read/subscribe multiple characteristics at a time.

zhuqingsong59, 2016-09-27:
Actually, I have a problem now,i can write successfully by characteristic which has attribute like this:"properties": {"write": true,  "writeWithoutResponse": true} '. so i try to get characteristic which has attribute like this:  "properties": { "read": true },use it's uuid for read(),and i set a timer for call read(),but i can't get data from bluetoothle devices,and i found all  characteristics that hava attribute like this:  "properties": { "read": true },it's descriptors of length is 0 ,Is this normal?,and i found a characteristic which has attribute like this:  "properties": {"notify": true},it's descriptors of length is not 0,so i use it's descriptorUUID to call readDescriptor(),i didn't get data  that i want either,soryy ,my english is poor, i hope you can understand what i was saying.i found litle information about how to use bluetoothle to 
Transmit data on Hybrid App,maybe i am in a wrong way?thanks alot

randdusing, 2016-09-29:
What's the error you get when you try to read a characteristic with the read property set to true?

zhuqingsong59, 2016-09-29:
i can execute readsucces function without error,when i sent data by a Bluetoothle module,it shows that i have send data Successfully,but i can't get the data that i sent before from readsuccess's returnobj.

zhuqingsong59, 2016-09-29:
now i can write data to my bluetoothle module,but i can't read data from my bluetoothle module.when i try to read a characteristic with the read property set to true,i get a complete object like your document,but object's value is a null value...my code like this:
for(var i=0;i<len;i++){
     serviceUUID = services[i].uuid;
     characteristics = services[i].characteristics;
      for(j in characteristics){
           characteristic=characteristics[j];
           var read= characteristic.properties.read || false;
           if(read){
              characteristicsUUID=characteristic.uuid;
               bluetoothle.read(readSuccess,readerror,{
                                address:app.address,
                                service:serviceUUID,
                                characteristic:characteristicsUUID
                })
           }
      }
 }
function readSuccess(obj){
    var bytes = bluetoothle.encodedStringToBytes(obj.value);
    var string = bluetoothle.bytesToString(bytes);
    $('.inform').append('<p>'+string+'....</p>');
}

randdusing, 2016-09-29:
Those values look correct, you need to convert that base64 encoded string into bytes using the helper function. ```#encodedStringToBytes()```

zhuqingsong59, 2016-09-29:
i have done it.but what ever i send from my bluetoothle device ,the read value is not change,just get three string "",' ','RmVhc3ljb21MRQ==' and "AAA=",use function convert to get '',"FeasycomLE",""

randdusing, 2016-09-30:
Are you sure the device is changing its read value?

zhuqingsong59, 2016-10-03:
I used a app from internet which is use  for testing bluethonle,it can get all data that i send from my bluetoothle module

zhuqingsong59, 2016-10-08:
ok,i am stupid.now i  have get data by subscribe() to subscribe a characteristic with the notify property set to true.but how to jude i should use read() or subscribe()?it denpend on bluetoothle device?or telephone?

randdusing, 2016-10-08:
It depends on the BLE peripheral device. Generally for data that doesn't change or changes very rarely, you use read. For stuff that changes frequently (like a heart beat), you use subscribe. But it really depends on how the BLE peripheral device is set up. The properties of the characteristic will show whether it supports reads, notification/indications or both.

zhuqingsong59, 2016-10-09:
thank for you help.

closed 2016-10-09 by zhuqingsong59