randdusing/cordova-plugin-bluetoothle #355: iOS now requires adding NSBluetoothPeripheralUsageDescription key to info.plist

mpedersen15, 2016-10-05:
It looks like a new requirement of iOS builds submitted to iTunes Connect is to include the  NSBluetoothPeripheralUsageDescription key with a valid text value in your info.plist file. 

Doing this manually in your info.plist file, allows me to upload builds successfully. I followed this [thread](https://groups.google.com/forum/#!msg/google-admob-ads-sdk/UmeVUDrcDaw/HIXR0kjUAgAJ): 

`<key>NSBluetoothPeripheralUsageDescription</key>`
`<string>Advertisement would like to use bluetooth.</string>`

Would it be an easy update to have the plugin modify the info.plist file automatically with the key and a generic text value?

randdusing, 2016-10-07:
Easy enough to add, but developers will have to change it every time they update the plugin if they decide to customize it. I may be able to make it a parameter that defaults to something generic. I'll try to take a look this weekend.

rgagne, 2017-03-06:
Any update on this or suggested workaround?

We're using Adobe Build, which prevents us from being able to edit the plist without forking the project (or something like that).

randdusing, 2017-03-07:
How about I just create another plugin that adds the plist? I don't want to overwrite changes that other users may have already made to the plist. I'm already doing something similar for the background modes. I may be able to make the actual text dynamic too. I believe I've seen another plugin that took parameters.

rgagne, 2017-03-07:
That would work.

randdusing, 2017-03-13:
@rgagne Check out https://github.com/randdusing/cordova-plugin-bluetooth-peripheral-usage-description. Let me know if you run into any issues.

alexhisen, 2017-06-26:
Won't the text be lost if the project is built on a different machine?

Since the text is supplied at the time of installing the plugin, it seems that it's only going to work once. If you reinstall ios platform or build the project on a different machine, you are not going to get the text, right?

I would imagine the best approach is to store the text as a configuration parameter in config.xml. This can be done with https://github.com/dpa99c/cordova-custom-config and the following in config.xml:
`    <config-file platform="ios" target="*-Info.plist" parent="NSBluetoothPeripheralUsageDescription">
        <string>This app requires Bluetooth to connect to our hardware.</string>
    </config-file>
`

randdusing, 2017-06-28:
I haven't been using Cordova lately, so I don't remember if it's a future of Cordova or Ionic. But I thought the plugin parameters were saved.

Mimateam, 2017-08-01:
I agree with @alexhisen

ricardoparro, 2017-11-20:
thanks @alexhisen!