randdusing/cordova-plugin-bluetoothle #356: not able to get a list of near by bluetooth devices.

funreadsapp, 2016-10-06:
Hello, 

i am having an issue with the plugin and was hoping to get some help with it. 
i put my min target API to 23, and made sure location is on.
but when i initialize, my status is always "enabled" and not "initialized", and when i scan i don't get back any devices (even though there are some when i look in the bluetooth section of the phone setting page).
any idea what else may be preventing me from seeing devices?
is there a place where there are more examples on how to use the plugin?

i even followed some of the code from this post https://github.com/randdusing/cordova-plugin-bluetoothle/issues/66 

any help would be greatly appreciated.

funreadsapp, 2016-10-06:
oh i realized i was confused and was toggling the OS level location and not on the app level. i am able to see come bt devices. 

has anyone tried connecting to raspberry pi? for some reason i dont see it when i scan but the phone will see it when i go to the bt manager page

randdusing, 2016-10-07:
Are you sure the raspberry pi is advertising as a BLE device?

funreadsapp, 2016-10-07:
you're absolutely right, thank you. i needed to put 

sudo hciconfig hci leadv 3 in the terminal of the pi.

closed 2016-10-08 by randdusing