randdusing/cordova-plugin-bluetoothle #359: bluetooth.close() doesn't really close the device in iOS 10 (Fine with 7/8/9)

jammind, 2016-10-08:
Before In iOS 7/8/9 (Working):
- When I call bluetoothle.close(), the device disconnects to my app. And it disappears from the iOS connected bluetooth device list.
- The LED on the device blinks again. And it appears in the list after I call bluetoothle.startScan().

Now In iOS 10 (Not working):
- When I call bluetoothle.close(), the device disconnects to my app. But it remains in the iOS connected bluetooth device list.
- The LED on the device remains on steady. And it doesn’t appear in the list after I call bluetoothle.startScan().
- I have to either switch the device off/on or turn iOS bluetooth off/on to make it work again.

randdusing, 2016-10-08:
What happens if you call disconnect() before calling close(). Generally you can just call close(), but maybe that has been changed in iOS10.

ikarus-balsero, 2016-10-08:
I experienced the same problem with iOS 10. The only workaround so far, as you mentioned, is calling disconnect() before close().

randdusing, 2016-10-08:
I use to enforce disconnecting before closing...so maybe I'll need to go back to that. Or at least recommend it for iOS 10.

jammind, 2016-10-08:
Calling disconnect() before close() does close the device. And next time before I call connect(), I need to call reconnect() first. Thanks for the tip. Please update your doc:)

randdusing, 2016-10-08:
Once you close() a device, you shouldn't be able to call reconnect()...only connect().

jammind, 2016-10-09:
In my case, no. I also found these quirks in iOS 10.

1) connect() to a device that I previously called disconnect() + close() will get the following error. Add a reconnect() before connect() will ensure connection with both NEW & OLD devices.

> address: "CD5418DA-4CE3-4358-83EF-2B8491FB1BEA"
> error: "connect"
> message: "Device previously connected, reconnect or close for new connection"
> name: "MB-o1-0053"
> 

2) bluetoothle.close(closeSuccess, closeError, params);
closeSuccess is never invoked.

3) bluetoothle.connect(connectSuccess, connectError, params);
connectSuccess is invoked only when it's established. Never invoked when it's disconnected. Instead I got an connectError call showing the following message.

> address: "CD5418DA-4CE3-4358-83EF-2B8491FB1BEA"
> error: "isDisconnected"
> message: "Device is disconnected"
> name: "MB-o1-0053"

randdusing, 2016-10-14:
That doesn't make any sense... Can you post all your code related to connecting and disconnecting?

t2wu, 2016-11-10:
Same issue here. Calling close doesn't cut it for iOS 10. It tells you the disconnect is successful, but it is in fact still connected. Calling disconnect prior to calling close will close it successfully.

sunny-mittal, 2016-11-23:
I've seen this issue as well and as my project is quite large and needs to work, I wrapped all the plugin calls in promises and always do this:

```
disconnect().catch(console.error)
  .then(close).catch(console.error)
  .then(/* anything else that needs to be done */)
```

I do that before every connect attempt and since I catch the case when a device may not be connected, it doesn't result in a crash. You'll just get a log saying "blah blah is not connected" or something.

NOTE: I partially apply all functions using a base function which pulls the address from a parent scope, hence it not being shown. The function is extremely simple:

```
        function simple(method, params) {
            return new Promise(function(resolve, reject) {
                var newParams = _.clone(params || {});
                if (!newParams.address) {
                    newParams.address = addr;
                }
                ble[method](function(res) {
                    resolve(extractVal(res));
                }, reject, newParams);
            });
        }
```

A simple use of this (using lodash) is:

`var disconnect = _.partial(simple, 'disconnect')`

It's true that any function that uses this internally will send an address that may or may not be needed, but thankfully this is JS and the presence of an unneeded argument is ignored. Hope this helps someone :)

sunny-mittal, 2016-11-23:
I should also note that I _never_ use `reconnect` per the docs' suggestion and have not had a problem. If you're finding yourself having to use it, there may be a different issue. Our support matrix covers iOS from 7 onward and Android from 4.3 onward, so it's fairly extensive, and no issues have been observed.

randdusing, 2016-11-24:
Perhaps I need to add promises to the plugin directly instead of relying on the Angular library...

jachstet-sea, 2018-06-18:
Is there a way to get the list removed from the list of Settings -> Bluetooth -> "My Devices" using this plugin on iOS 10? Even if I call disconnect() and close(), the device is still in that list and the PIN for the device is not needed to connect to the device.