randdusing/cordova-plugin-bluetoothle #360: bluetoothle.startScan()

Aaronwong94, 2016-10-10:
I had a question on the startScan(), where i use this function in a "refreshDeviceList" button. it seems like, at first i open the application, the first time i press the button and it successfull scan the bluetooth device .However when i press for the second time on the button, it return nothing! and go to error callback. 
The question is 
1. Is the startScan() only return once ? because when i press second time on "refreshDeviceList" button it go to onError callback. 
2. or it should be stopScan() before another startScan()?
btw, it seem like the "timeout:5000" i add in the statScan(param) does work .
In my application , the disconnect(showMainpage,error), the success callback will return to the main page where it had the device list and the "refreshDeviceList"button, the device list will content duplicate of  device scanned on the list which indicate the scan are not stop however i had put the timeout in the statScan(param). 
 I was new in app development! thx for the help!^^
code:
refreshDeviceList: function (result) {
            var resultStatus = result.target.dataset.results;
           var platform = window.cordova.platformId;
            alert("Refresh list success!");
            deviceList.innerHTML = ''; // empties the list
            if (resultStatus == "enabled") {
                if (platform == 'android') {
                    //alert("This is android platform");// Android filtering is broken
                    bluetoothle.startScan(app.onDiscoverDevice, app.onError, { services: [],timeout:5000});
                }}},
onDiscoverDevice: function (device) {
            //deviceList.innerHTML = "";
            var oldDevice = device.address;
            if (device.name != oldDevice) {
                var listItem = document.createElement('li'),
                html = '<b>' + device.name + '</b><br/>' +
                    'RSSI: ' + device.rssi + '&nbsp;|&nbsp;' +
                    device.address;
            listItem.dataset.deviceId = device.address;
            listItem.dataset.devicename = device.name;
            listItem.dataset.devicerssi = device.rssi;
            listItem.dataset.devicestatus = device.status;
            disconnectButton.dataset.deviceId = device.address;
            listItem.innerHTML = html;
            deviceList.appendChild(listItem);
            }
            else {
                deviceList.innerHTML = "";
            } },

randdusing, 2016-10-14:
What type of error are you receiving? Most likely it's trying to start the scan again while the scan is already in progress. Are you on Android 6.0? Make sure your app has the proper permission as explained in the readme.

Also check out this guide to markdown: https://guides.github.com/features/mastering-markdown/ Formatting your code will make it easier to receive help!

mstrayer, 2017-02-24:
I am getting behavior similar to this.  Fresh device start, open the app, scan, no problem.  Re-run the app and start scan doesn't return any devices.  Occasionally I get an error code of 5 from the start scan which is unfortunate because the documented error codes only go to 4 as far as I can tell.  Once this state has occurred the only way out of it is to cycle the bluetooth on and off at the OS level.  I am filtering based on a service UUID and only allowing the first match if that would matter.