randdusing/cordova-plugin-bluetoothle #361: subscribe multiple

screengarden, 2016-10-10:
hi,

i'm new to bluetoothle. how can i subscribe to multiple characteristics?

if i write the following code, my app is doing nothing. if i write just one block (one subscription) everthing is fine.

please help me.
`
    bluetoothle.subscribe(function (result) {
        if (result.value) {
            var ergebniss = app.convertBytearray2dec(result.value);
            document.querySelector('#TXT_AKKU_LOW').value = ergebniss;
        }
    }, function (e) {
        console.log(e);
        console.log("subscribe fehler");
    }, {
        "address": app.deviceAddress,
        "service": app.werte.OWNSERVICE,
        "characteristic": app.werte.AKKU_LOW
    });

  bluetoothle.subscribe(function (result) {
        if (result.value) {
            var ergebniss = app.convertBytearray2dec(result.value);
            document.querySelector('#TXT_ZUGKRFAT_LOW').value = ergebniss;
        }
    }, function (e) {
        console.log(e);
        console.log("subscribe fehler");
    }, {
        "address": app.deviceAddress,
        "service": app.werte.OWNSERVICE,
        "characteristic": app.werte.ZUGKRFAT_LOW
    });

    bluetoothle.subscribe(function (result) {
        if (result.value) {
            var ergebniss = app.convertBytearray2dec(result.value);
            document.querySelector('#TXT_ZUGKRFAT_HIGH').value = ergebniss;
        }
    }, function (e) {
        console.log(e);
        console.log("subscribe fehler");
    }, {
        "address": app.deviceAddress,
        "service": app.werte.OWNSERVICE,
        "characteristic": app.werte.ZUGKRFAT_HIGH
    });
    
    
 `

hannibaldinski, 2016-10-12:
Yes, it should be possible. I've used it once but thats some time ago. Though I used a subscriptionHandler function as callback for each subscription. Maybe try something like this?

bluetoothle.subscribe(subscribeHandler, errorHandler, {
    "address": app.deviceAddress,
    "service": app.werte.OWNSERVICE,
    "characteristic": app.werte.ZUGKRFAT_HIGH
});

function subscribeHandler(result) {  
   if (result.service === app.werte.OWNSERVICE && result.characteristic ===   app.werte.ZUGKRFAT_HIGH) {
       // do stuff
  }

  if (result.service === app.werte.OWNSERVICE && result.characteristic === app.werte.AKKU_LOW) {
      // do stuff
  }
}

randdusing, 2016-10-14:
That looks perfectly fine. Do you see any runtime errors? Maybe add a console.log to the success handler to double check.