randdusing/cordova-plugin-bluetoothle #365: Writes seem to get corrupted or not sent completely

JacobYoungDigi, 2016-10-13:
After successfully discovering the device, I attempt to send multiple message to one of the characteristics. But something bad seems to be happening on the success callback. What appears to be happening is that the full message is not even being transmitted over the air. Here are some log messages:

Sending request data: 66,2,16,0,115,105,111,110,61,34,49,46,49,34,62,60,113,117,101,114 (these are in bytes)
Write success: 66,2,16,0,0

As you can see, on the success callback, most of the bytes are just getting replaced with a 0. This is the code that creates the byte string:

`var writeSuccess = function (obj) {
            console.log("Write success: " + bluetoothle.encodedStringToBytes(obj.value));
};
var writeError = function (obj) {
            console.log("Write error: " + obj.message);
};
rci_request = '<rci_request version="1.1"><query_state><device_info /></query_state></rci_request>';
var packer = new JSPack();
var request = packer.Pack("<BBH", [0x42, 2, 16]);
request = request.concat(bluetoothle.stringToBytes(rci_request.substring(16,32)));
console.log("Sending request data: " + request);
var encodedString = bluetoothle.bytesToEncodedString(request);
var params = {
            "address": address, 
            "service": service,
            "characteristic": request,
            "value": encodedString
};
bluetoothle.write(writeSuccess, writeError, params);`

And it does not seem to be the device that is the problem, as I have tested doing these same commands over BLE in Android Studio.

Also, this is how I am initializing the bluetooth:

`bluetoothle.initialize(initializeSuccess, {
        "request": true,
        "statusReceiver": false,
        "restoreKey": "bluetoothleplugin"
});`

And finally, this is the JS I am using for the `packer.Pack("<BBH", [0x42, 2, 16]);` line: 
https://github.com/pgriess/node-jspack/blob/master/jspack.js

Thanks.

JacobYoungDigi, 2016-10-13:
HOWEVER, before I do a write like that, I send this message that is successful, following the same process as above:

var request = packer.Pack("<BBH", [0x42, 0, 83]);
console.log("Sending request data: " + request);
var encodedString = bluetoothle.bytesToEncodedString(request);
var params = {
"address": address, 
"service": service,
"characteristic": request,
"value": encodedString
};
bluetoothle.write(writeSuccess, writeError, params);

Sending request data: 66,0,83,0
Write success: 66,0,83,0

randdusing, 2016-10-14:
I'm not really sure how packer.Pack works, so that me be contributing to the problem. Can you try writing an increasing amount of bytes all the way up to 23ish and see where it begins to fail? Like write [0], [0,1] ... [0, ..., 22] etc. If this works, probably an issue with how packer.Pack works. If this doesn't work, we'll keep digging. Also, can you print out the encodedString for both requests?

mikewadsten, 2016-10-14:
I'm actually fairly sure the issue is with this line:

```
request = request.concat(bluetoothle.stringToBytes(rci_request.substring(16,32)));
```

`stringToBytes` returns a Uint8Array.

`[1, 2].concat(new Uint8Array([3, 4]))` returns `[1, 2, <the Uint8Array>]`.

`String.fromCharCode(new Uint8Array([1, 2]))` (as called in [`bytesToEncodedString`](https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/www/bluetoothle.js#L151)) returns `'\u0000'`.

So, I think `request.concat([].slice.call(bluetoothle.stringToBytes(rci_request.substring(16, 32))))` will do what is expected.

JacobYoungDigi, 2016-10-14:
Yes, Mike is correct. The writes seem to work correctly now. Thank you!

randdusing, 2016-10-15:
Cool glad it works now.

closed 2016-10-15 by randdusing