randdusing/cordova-plugin-bluetoothle #367: It's possible to do a sequence of actions on background?

DaniloCouto, 2016-10-20:
I want to scan, connect, and write in background. Is it possible?
Which platforms? Just on IOS it's possible?
On Android is needed to use another plugin? 
Or need to combine another plugin like cordova-plugin-background-mode?

I have these questions because I can’t find anywhere if android is possible to do such things on background mode.

Thanks for your attention.

randdusing, 2016-10-21:
On iOS, you just need to install the bluetooth background mode plugin that's mentioned in the documentation. Even if in the background, your app will 'awake' for ~10 seconds whenever a new device is discovered, connected, read, subscribed, etc.

On Android, you need to use that plugin. Don't use the plugin on iOS or your app may not be approved!

DaniloCouto, 2016-10-25:
Thanks, this helps a lot o/
Some libs don't add this behavior as a default.
Consider this #367 closed.

closed 2017-06-29 by DaniloCouto