randdusing/cordova-plugin-bluetoothle #368: How to wait for read/write success/failure before moving on

JacobYoungDigi, 2016-10-21:
This is less of an issue but more of a javascript question. This is my code (assume params is correctly set):

var writeRequest = function (parameters) {
        var writeSuccess = function (obj) {
            console.log("Write success");
        };

        var writeError = function (obj) {
            console.log("Write error");
        };
        bluetoothle.write(writeSuccess, writeError, parameters);
};
writeRequest(params);
console.log("write complete");

How do I correctly wait for either "Write success" or "Write error" to be printed before "write complete"?
I know I have to use something like Promises, but I am having trouble (I am fairly new to async requests and JS) and hope that someone could help. Thanks!

JacobYoungDigi, 2016-10-21:
Nevermind I think I figured it out.

closed 2016-10-21 by JacobYoungDigi

randdusing, 2016-10-21:
Glad you figured it out. Currently only the Angular wrapper supports promises. I really need to make promises optional directly with the plugin Javascript. The tricky part is native promises don't support a third notify callback like Angular's $q does, which is used by connect, subscribe, etc.

I'm still trying to figure out the best way to support them. Probably just need to pass in a 3rd callback that will only return while the promise is still pending.

I highly recommend using promises in some way. Bluetooth LE is very asynchronous especially when going through a Cordova plugin. I can't imagine writing a Bluetooth LE app without the help of promises... which is one reason I removed the 'native' Javascript example app and put an Angular based one in the Angular wrapper.