randdusing/cordova-plugin-bluetoothle #369: Can't get any Devices

jmd90, 2016-10-24:
Hello Everybody,

I'm trying to use the Bluetooth Plugin but i can't get any Devices by starting the scan mode. I'm Using TS with AngularJS. I have a Service UUID of an smartglass which I want to reach.

Maybe somebody can give me a hint?

Here's my Code Snipped:

`class InfoViewController {
        public version: string;
        public build: string;
        public model: string;
        public platform: string;
        public platformVersion: string;

        public currentUser: IUser;
        public baseUrl: string;
        public language: string;

        static $inject = ['tools', 'bluetoothService'];
        constructor(
            tools: ToolsService,
            bluetooth: plugin.IBluetoothService

        ) {
            const cordovaDevice: Device = ionic.Platform.device();
            this.version = !!cordovaDevice.platform ? tools.getAppVersion() : '4.0.0';  // <<< fallback for BROWSER
            this.build = BUILD_VERSION;
            this.model = !!cordovaDevice.model ? cordovaDevice.model : 'BROWSER';
            this.platform = !!cordovaDevice.platform ? cordovaDevice.platform : 'BROWSER';
            this.platformVersion = !!cordovaDevice.version ? cordovaDevice.version : '';

            this.currentUser = tools.currentUser;
            this.baseUrl = this.currentUser.BaseUrl;
            this.language = this.currentUser.LanguageCode;

            this.initBluetooth(bluetooth);

        }

        initBluetooth = (bluetooth: plugin.IBluetoothService) => {

            let address = "98:DA:92:00:1A:B8";
            const k: InitializationParams = {
                request: true,
                statusReceiver: true,
                restoreKey: 'test'
            };

            bluetooth.initialize(k).then(success => {
                console.log(success);
            }).then(x => {

                return bluetooth.isLocationEnabled()
            }).then(success => {
                console.log(success)
                return bluetooth.hasPermission()
            }).then(permission => {
                console.log(permission)
                return bluetooth.startScan({
                    services: [
                        "3D0F22C3-E3B5-4469-A5D4-98F459C6E531",
                        "46ABB8F6-FEE4-45B1-AAB2-DAB93DEE8CB1"
                    ],
                    scanMode: 2,
                    matchMode: 1,
                    matchNum: 3,
                    callbackType: 1
                })
                }).then((erg) => {
                    console.log(erg);
                    return bluetooth.retrieveConnected({ services: [] });
                }).then((erg) => {
                    console.log(erg);
                }).catch(reject => {
                console.log(reject);
            });
        }

    }

    angular
        .module('mfapp.infoview')
    `

randdusing, 2016-10-24:
startScan uses notify to return scan results. It looks like you're only using the resolve and reject. Check out the angular wrapper example.

jmd90, 2016-10-24:
I think I don't get it what you are trying to say me. I'm using the two options to receive an success or failure with promises. Do I have to implement an extra function to listen on events for example if a new device appeared?

randdusing, 2016-10-25:
A promise will only resolve once and the scan callback is called multiple times.

jmd90, 2016-10-25:
Okay that makes sense. I'm trying to rewrite my Code and give feedback soon. 

A question to the lifecycles. There is a Central Life Cycle and a Peripheral Life Cycle. I don't get the idea whats the difference between them or how they interact with each other. 

Another question is if i can retrieve connected devices which i connected outside of my app. 
For example: 
Pair devices outside of the  app
init the plugin
retrieveConnected -> get a List of paired device?

Would be nice to get some help of you.

Thanks for your fast reply

JMD

randdusing, 2016-10-26:
I recommend always using retrieveConnected and then startScan to find devices. The retrieveConnected behavior is a bit strange depending on operating system and version.

Central Life Cycle - The phone is connecting to a device like a heart rate monitor
Peripheral Life Cycle - Your phone is advertising as a device. Other phones may connect to it.

jmd90, 2016-10-26:
Thank you very much. I think the issue can be closed. The device i was searching had the Android API 15 and it looks like it doesn't support Bluetooth Low Energy.

closed 2016-10-26 by jmd90

randdusing, 2016-10-26:
Just to clarify. Even if the Android device was brand new and supported BLE, you probably wouldn't see it during a scan...unless an app was causing it to actively advertise as a BLE device.