randdusing/cordova-plugin-bluetoothle #370: Write method do nothing (iOS)

dannygunawan, 2016-10-25:
Hello,
I'm trying to print to a Bluetooth Printer from an iOS 10 device.

```
bluetoothle.connect(function(r){
  console.log(r)
}, failure,
{
  "address":"73CB49C5-E5C8-407D-BBCE-E3063D9F3262"
});
```

return:
`{status: "connected", name: "BlueTooth Printer", address: "73CB49C5-E5C8-407D-BBCE-E3063D9F3262"} 
`
then:
```
var string = "Write Hello World";
var bytes = bluetoothle.stringToBytes(string);
var encodedString = bluetoothle.bytesToEncodedString(bytes);
bluetoothle.write(function(r){
  console.log(r)
},failure,
{
  "service": "E7810A71-73AE-499D-8C15-FAA9AEF0C3F2", 
  "address":"73CB49C5-E5C8-407D-BBCE-E3063D9F3262",
  "characteristic":"BEF8D6C9-9C21-4C9E-B632-BD58C1009F9F",
  "value":encodedString,
  "type":"noResponse"}
);
```
result:
```
{
address: "73CB49C5-E5C8-407D-BBCE-E3063D9F3262",
characteristic: "BEF8D6C9-9C21-4C9E-B632-BD58C1009F9F",
name: "BlueTooth Printer",
service: "E7810A71-73AE-499D-8C15-FAA9AEF0C3F2",
status: "written",
value: "V3JpdGUgSGVsbG8gV29ybGQ="
}
```

However, nothing is printed. Any clue? Is there something that I missed? Thank you.

randdusing, 2016-10-26:
It seems like the write was successful according to iOS. So you'll probably need to do more debugging on the printer side. That's a really short message. Are you sure you don't need to send more data?

Also, discover and write shouldn't run sequentially. Write should only be called after discover has successfully completed.

dannygunawan, 2016-10-26:
hi @randdusing,
Thank for your reply. After testing on both Android and IOS, I found that the problem was on the message. I forgot to add ESC/POS code bytes in front of the message.

closed 2016-10-26 by dannygunawan

maxdimaz, 2017-09-06:
How to add ESC/POS code byte?