randdusing/cordova-plugin-bluetoothle #373: Error Domain Code=10 on subscribe iOS 10

LucasChapuis, 2016-10-31:
Hi,
I saw some other people also had the same issue but I didn't succeed fixing it.
I've got no problem on Android but I get in the error function for iOS, I get the right service and characteristic in the callback but an error message : 

```
error: "subscription"
message: "Error Domain=CBATTErrorDomain Code=10 \"The attribute could not be found.\" UserInfo={NSLocalizedDescription=The attribute could not be found. »
```

I've no ideas how to fix it, I've found this http://lists.apple.com/archives/bluetooth-dev/2013/Aug/msg00023.html, but it didn't do a thing.

Any idea ?

LucasChapuis, 2016-11-06:
Still having the issue after a week ... I'm desperate^^, still no idea ?

randdusing, 2016-11-06:
Can you list out the result of discover or services/characteristics/descriptors?

LucasChapuis, 2016-11-07:
I'm using discover for iOS and the result is : 

```
{
  address: "B83737DD-C287-AC74-42F9-7478E1AC717F",
  name: "In_Et_Motion",
  services: [
    {
      characteristics: [
        {uuid: "496E264D-6F74-696F-6E01-010000000000", properties: {indicate: true, notify: true}, descriptors: []}
      ],
      uuid: "496E264D-6F74-696F-6E01-000000000000"
    },
    {
      characteristics: [
        {uuid: "496E264D-6F74-696F-6E01-110000000000", properties: {indicate: true, notify: true}, descriptors: []}
      ],
      uuid: "496E264D-6F74-696F-6E01-100000000000"
    },
    {
      characteristics: [
        {uuid: "496E264D-6F74-696F-6E01-210000000000", properties: {indicate: true, notify: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-220000000000", properties: {indicate: true, notify: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-230000000000", properties: {indicate: true, notify: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-240000000000", properties: {indicate: true, notify: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-250000000000", properties: {indicate: true, notify: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-260000000000", properties: {indicate: true, notify: true}, descriptors: []}
      ],
      uuid: "496E264D-6F74-696F-6E01-200000000000"
    },
    {
      characteristics: [
        {uuid: "496E264D-6F74-696F-6E01-310000000000", properties: {indicate: true, notify: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-320000000000", properties: {write: true, read: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-330000000000", properties: {write: true, read: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-340000000000", properties: {write: true, read: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-350000000000", properties: {write: true, read: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-370000000000", properties: {write: true, read: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-360000000000", properties: {read: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-380000000000", properties: {notify: true, indicate: true, read: true}, descriptors: []}
      ],
      uuid: "496E264D-6F74-696F-6E01-300000000000"
    },
    {
      characteristics: [
        {uuid: "496E264D-6F74-696F-6E01-410000000000", properties: {write: true}, descriptors: []}
      ],
      uuid: "496E264D-6F74-696F-6E01-400000000000"
    },
    {
      characteristics: [
        {uuid: "2A25", properties: {read: true}, descriptors: []},
        {uuid: "2A29", properties: {read: true}, descriptors: []},
        {uuid: "2A24", properties: {read: true}, descriptors: []},
        {uuid: "2A27", properties: {read: true}, descriptors: []},
        {uuid: "2A26", properties: {read: true}, descriptors: []},
        {uuid: "2A28", properties: {read: true}, descriptors: []}
      ],
      uuid: "180A"
    },
    {
      characteristics: [
        {uuid: "496E264D-6F74-696F-6E01-510000000000", properties: {indicate: true, notify: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-520000000000", properties: {indicate: true, notify: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-530000000000", properties: {indicate: true, notify: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-540000000000", properties: {write: true, read: true}, descriptors: []},
        {uuid: "496E264D-6F74-696F-6E01-550000000000", properties: {write: true, read: true}, descriptors: []}
      ],
      uuid: "496E264D-6F74-696F-6E01-500000000000"
    },
    {
      characteristics: [
        {uuid: "E3DD50BF-F7A7-4E99-838E-570A086C666B", properties: {indicate: true, write: true, notify: true}, descriptors: [{uuid: "2902"}]},
        {uuid: "92E86C7A-D961-4091-B74F-2409E72EFE36", properties: {write: true}, descriptors: []},
        {uuid: "347F7608-2E2D-47EB-913B-75D4EDC4DE3B", properties: {read: true}, descriptors: []}
      ],
      uuid: "9E5D1E47-5C13-43A0-8635-82AD38A1386F"
    }
  ],
  status: "discovered"
}
```

I'm trying to subscribe to characteristic ```496E264D-6F74-696F-6E01-520000000000``` of service ```496E264D-6F74-696F-6E01-500000000000```

LucasChapuis, 2016-11-07:
Before trying to subscribe, I write a code in the characteristic `496E264D-6F74-696F-6E01-410000000000` of service `496E264D-6F74-696F-6E01-400000000000` and it works, so I don't think the problem comes from the discover ... otherwise I couldn't write

randdusing, 2016-11-08:
496E264D-6F74-696F-6E01-410000000000 doesn't have indicate or notify properties, so it can't be subscribed.

LucasChapuis, 2016-11-08:
No no no, I mean, I Write something in `496E264D-6F74-696F-6E01-410000000000`, It works, then I try to subscribe to `496E264D-6F74-696F-6E01-520000000000`, and I've got the error, even if the characteristic is `{indicate: true, notify: true}`

randdusing, 2016-11-09:
What happens if you don't write and just subscribe initially? Possibly something is changing on the device side when the write is triggered.

LucasChapuis, 2016-11-09:
I actually can't, this Write is a "pinCode", that makes me staying connected, if I don't write it, I don't get any data from the BLE device.
If I directly subscribe I wont get any data without this pinCode.

I tried anyway to directly subscribe, whitout writing anything, still got the same issue, same error message on subscribe callback

randdusing, 2016-11-11:
Do you control the device code? Maybe something is wrong on its side.

LucasChapuis, 2016-11-21:
Yeah I can change it, but I've got no reason to change it, I just tip it to get permission to read the datas on the card. Strange thing is that it's the same code for iOS et Android and everything is working well on Android...

randdusing, 2016-11-24:
I would try to remove all other variables and see if iOS and Android can both subscribe to a characteristic without worrying about any other functionality.

LucasChapuis, 2016-11-24:
After a bunch of tries, it's seems it came from the card where the
characteristics were badly implemented, like they had something missing,
that's why iOS didn't recognize them, Android had no problem with it, iOS
blocked it.

I tried to subscribe to another type of card (SPI) and it worked on iOS and
Android, so the plugin works, it was the card which had a problem.

>
Interesting thing is that Android didnt block it even if the characteristic
was not well implemented.

randdusing, 2016-11-24:
Android seems to be a little less stringent but pretty buggy. Where as iOS is pretty strict and tends to be pretty robust. Perhaps there is some sort of correlation there :)

weihuaronaldo, 2018-03-05:
Hi @LucasChapuis, we have met the same problem. Our app failed to set notification while the characteristics descriptor missing. But in case,  the problem comes when we use the newest 5th generation ipad with LightBlue, while other devices (such as iphone6s, ipad mini 2) is OK. Any ideas to solve it?

hardikdarji, 2019-10-11:
Hi I have the same issue with iOS for Bosch XDK device
for service with UUID: 5A211D40-7166-11E4-82F8-0800200C9A66 is XDK-Accelerometer
And below is characteristics:
Accelerometer    X-Axis    int16_t    5a211d41-7166-11e4-82f8-0800200c9a66
Accelerometer    Y-Axis    int16_t    5a211d42-7166-11e4-82f8-0800200c9a66
Accelerometer    Z-Axis    int16_t    5a211d43-7166-11e4-82f8-0800200c9a66

I am trying to notify to get x,y,z characteristics but get en error:

Error Domain=CBATTErrorDomain Code=10 "The attribute could not be found." UserInfo={NSLocalizedDescription=The attribute could not be found.}
Notification can't begin on characteristic

Any help for how to solve this?