randdusing/cordova-plugin-bluetoothle #374: clearCache Hanging

unclespode, 2016-11-03:
I'm in the process of doing a DFU update. When I trigger my device into bootloader mode, the services change, but the discovery does not - it still reports the old services. Using clearCache is obviously what is required so that it refreshes the services.

However, when I try and do this after a fresh scan and connection, it will simply hang on discovery. No error/success callbacks are called. If I try and run discovery again, it says that discovery is already in process. I have to terminate the app and try again for it to work.

unclespode, 2016-11-04:
`bluetoothle.discover(function(e) {
            console.log("Discovered", e);
            callback(true);
        }, function(e) {
            callback(false, e.message);
        }, {
            address: myAddress,
            clearCache : true
        });`

It works fine, if I don't have the clearCache parameter. As soon as it's in there, that's when the problems start.

randdusing, 2016-11-08:
Are you on 7.0? It's hanging up for me as well. But discovery is pretty messed up either way...

unclespode, 2016-11-08:
No, 6.

kgillespieanaren, 2017-02-17:
We are running into the issue as well. When we try to use the clearCache set to true.

I can get you a list of the devices we test and what versions of android they have and which once are running into this issue in a bit.

kgillespieanaren, 2017-02-17:
It appears that you would want to use the clearCache function right after you call connectGatt. So would this work if we passed the clearCache function to the connect function and call it right after connectGatt. See this forum for some details.

http://stackoverflow.com/questions/22596951/how-to-programmatically-force-bluetooth-low-energy-service-discovery-on-android

It may also be because this isn't a function that is actually exposed via the Android API so there is no guarantee that it'll keep working between versions of the android releases.