randdusing/cordova-plugin-bluetoothle #376: callback the function" bluetoothle.subscribe()"to recived the value to parse ,but the data is wrong in ios！

change1987, 2016-11-08:
var data = "";
function subscrbeCharac(){ 
  bluetoothle.subscribe(function(result){                        
                        if(result.status=="subscribed"){
                        alert("sucessful！"+"\n");
                        }
                        else if(result.status=="subscribedResult"){
                        var gyroscopeValueBytes = bluetoothle.encodedStringToBytes(result.value);
                        var gyroscopeValueString = bluetoothle.bytesToString(gyroscopeValueBytes);
                        data += gyroscopeValueString;
                        alert("data:"+data+"\n");
                        }    
                        },
                        function(error){
                        alert("failed!"+error.error);
                        },
                        {address:address,
                        service:"FFE0",
                        characteristic:"FFE1"});
  }
when the callback execute five times ,the received data is wrong with the format from the device !
if i delete the " alert("data:"+data+"\n");",the received data is correct!
How can i solve the problem?

randdusing, 2016-11-09:
I don't see how that's possible... ```alert()``` shouldn't be modifying data. But it sounds like you already know how to solve your problem, don't use ```alert()``` :) ! Try using ```console.log()``` instead.

change1987, 2016-11-10:
it's can't be work that add other code to deal the result.value !When i using code to parse the data from the callback  value ,the received from the device is wrong because the variable data no in timing sequence to add value.it's disorder. The sequence don't work in the order. I doubt the reason it's synchronous and asynchronous.The same code work in android that no problem,but in ios is always wrong.

change1987, 2016-11-10:
The device send the 160 bytes ,so the callback execute many times.And ,the received data from timing sequence  maybe disorder.The received data length is correct,but the parse data format is wrong.And the bluetooth plugin code in ios using the  asynchronous process to execute the code,so the mix order create. How can i do ?

change1987, 2016-11-10:
thanks ,the reason it's the alert() function .

randdusing, 2016-11-10:
It's working for you now?

change1987, 2016-11-16:
yes,the alert() function disorder the timing sequence to send data.

randdusing, 2016-11-17:
Cool, reopen if you run into any more issues.

closed 2016-11-17 by randdusing