randdusing/cordova-plugin-bluetoothle #378: bluetoothle.writefunction has error: Service not found

Aaronwong94, 2016-11-10:
I'm using bluetoothle.write(). the {para} i had hardcode with the service uuid and charcteristic uuid (since I know the service uuid and charcteristic uuid) however the error i obtain said the Service not found. Please help!!

Huiting-Yeong, 2016-11-10:
Had a similar error when trying to subscribe to a service. I solved it by running bluetoothle.discover() first then subscribing. You can try discovering first then write.

randdusing, 2016-11-10:
Yep, you need to discover prior to using any actions

Aaronwong94, 2016-11-10:
I had try using bluetoothle.discover(). I can get the device.adress, but can't get the serviceuuid and characteristicuuid, how should i retrieve that ?As what i get device.services are an object array,

randdusing, 2016-11-11:
You need to parse the object if you want to iterate through the services and characteristics. The format is documented in the readme. But you don't necessarily need to do that, just discover and then call read/write/etc

Aaronwong94, 2016-11-14:
Sorry I don't quit get it Cus when I dicover how can I retrieve the service uuid and characteristics uuid Cus what I get is a object array. What should I do to get the uuid for read and write function {param}

randdusing, 2016-11-14:
Take another look at the documentation and check out the Angular wrapper provided in the example if needed.

closed 2016-11-14 by randdusing

akashkhairnar01, 2020-06-06:
where is angular wrapper provided example

randdusing, 2020-06-06:
https://github.com/randdusing/ng-cordova-bluetoothle