randdusing/cordova-plugin-bluetoothle #379: bluetoothle.write : Unable to write on return

Huiting-Yeong, 2016-11-10:
Hi

I am trying to write a value to the ble object. I am successful in initialize, startScan, connect, discover and subscribe. 

When I run bluetoothle.write(), I am getting an error: **Unable to write on return**

```
var string = "write value";
var bytes = bluetoothle.stringToBytes(string);
var encodedString = bluetoothle.bytesToEncodedString(bytes);

new Promise(function (resolve, reject) {

  bluetoothle.write(resolve, reject, { "address" : zoneA.macAddress, "service" : "ffe5", "characteristic" : "ffe9", "value" : encodedString});

}).then(writeSuccess, handleError);

function writeSuccess(result){
  console.log(result);
}

function handleError(error){
  console.log(error);
}

```

Error:

> address: "..."characteristic: "FFE9"error: "write"message: "Unable to write on return"name: "..."service: "FFE5"

Any idea on what I am doing wrong?

randdusing, 2016-11-10:
That looks fine to me. Can you list out the properties for that service that are in the discover call?

Huiting-Yeong, 2016-11-11:
[image]

This is the list of properties I am getting

Huiting-Yeong, 2016-11-11:
Did some debugging in the java file and realised in `onCharacteristicWrite(...)`, the status value is equal to 133 (BluetoothGatt.GATT_ERROR). Any idea how does that come about?

randdusing, 2016-11-11:
Can you get the system level Bluetooth HCI logs or some higher level BLE logs from Android Device Monitor? And what type of phone is this on?

Huiting-Yeong, 2016-11-16:
Yes, I am able to get Bluetooth HCI logs and have tried opening them with wireshark. But I am not sure how to interpret them. I am running it on Android 5.0.

Huiting-Yeong, 2016-11-16:
Hi, it is suddenly returning me a status of 0, even though I didn't change any code. Not sure why but thanks so much for all your help!

closed 2016-11-16 by Huiting-Yeong

randdusing, 2016-11-17:
No worries. Maybe a fluke on the phone or device side, but glad it's working now.

Funkelfetisch, 2021-08-19:
I'm having the same problem (I think)

ZweiEuro, 2021-09-06:
I am having the same issue

clarklight, 2021-09-14:
I am having the same problem as well. when i do .read() it says unable to read, when i do write() it says Unable to write on return, i am using another android phone and setting it as a peripheral.