randdusing/cordova-plugin-bluetoothle #380: "Application registration failed" Error

AndreiChe, 2016-11-14:
Hello!

One our client can't scan BLE devices because of the error: "Application registration failed". I've tried to investigate this problem and found that it is an Android issue. There are many StackOverflow posts like this: http://stackoverflow.com/questions/35376682/how-to-fix-android-ble-scan-failed-application-registration-failed-error. Could please apply this fix? Or should I prepare merge request (I'm not an Android developer)?

To solve the problem our client (Android 6.0.1, SAMSUNG-SM-N920A) tried this:
* Restart phone
* Reinstall the app

PS: Thank you for you work. Great plugin!

randdusing, 2016-11-14:
What do your scan parameters look like? You may try disabling and re-enabling Bluetooth programmatically if you detect this error, but that's a bit hacky.

AndreiChe, 2016-11-14:
Thank you for quick reply. I use only one parameter for scan:
```{
services: ['180d']
}```

Do you mean to use something like this?
```
bluetoothle.disable(function(){
  bluetoothle.enable()
});
```

Is this an equivalent of the code?
```
BluetoothAdapter.getDefaultAdapter().disable();
...
case BluetoothAdapter.STATE_OFF:
  Log.d(TAG, "bluetooth adapter turned off");
  handler.postDelayed(new Runnable() {
    @Override
    public void run() {
        Log.d(TAG, "bluetooth adapter try to enable");
        BluetoothAdapter.getDefaultAdapter().enable();
    }}, 500);
  break;
```

randdusing, 2016-11-17:
Scan looks fine. Yep, it's basically doing that although maybe a little bit quicker than the 500ms timeout. Android support added some new scan parameters, so maybe the default scan parameters are causing issues with that particular phone. You could try specifying the properties to override the defaults.

RitterD, 2018-06-21:
Hello randdusing,

I always get the Error: "Error on startScan: Application registration failed0 error"
It occurs after I want to scan....

Here the full console.log:
startScanSuccess(scanStarted)
Scanning for devices (will continue to scan until you select a device)... status
Error on startScan: Application registration failed0 error

@AndreiChe did the disable and enable work for you? I tried it out but it only disables bluetooth and the error occurs any way.... can you tell me at whitch point you placed the disable and re-enable function?

Thank you very much in advance!!

Great documentation btw =)

`var foundDevices=[];

function startscan(){

    foundDevices=[];

    document.getElementById("devices").innerHTML = "";
    document.getElementById("services").innerHTML = "";
    document.getElementById("output").innerHTML = "";

    bluetoothle.startScan(startScanSuccess, handleError, {

        services:[]
        //allowDuplicates: true,
        //scanMode: bluetoothle.SCAN_MODE_LOW_LATENCY,
        //matchMode: bluetoothle.MATCH_MODE_AGGRESSIVE,
        //matchNum: bluetoothle.MATCH_NUM_MAX_ADVERTISEMENT,
        //callbackType: bluetoothle.CALLBACK_TYPE_ALL_MATCHES
    });

}
    function startScanSuccess(result) {

        //alert("StartScanSuccess " + result.status);

        console.log("startScanSuccess(" + result.status + ")");

        if (result.status === "scanStarted") {

            console.log("Scanning for devices (will continue to scan until you select a device)...", "status");
        }
        else if (result.status === "scanResult") {

            if (!foundDevices.some(function (device) {

                return device.address === result.address;

            })) {

                console.log('FOUND DEVICE:');
                console.log(result);
                foundDevices.push(result);
                addDevice(result.name, result.address);
            }
        }

    }`

randdusing, 2018-06-23:
Does the angular example app work for you? I'd try this out: initialize -> disable -> enable -> scan. Wait until each action is complete until moving to the next.

ragcsalo, 2018-08-17:
You get the error "Application registration failed" when the app does not have the necessary permissions enabled (like LOCATION)!!! I had the same issue, and enabling the permission solved the problem. :-)

HansKrywaa, 2019-10-28:
@randdusing I have the same Issue. I already tried disabled -> enabled -> startScan, but it doesn't work. Also i have enabled everything i need on my Devie (Bluetooth, Location, etc.).

What could it else be?

ebk46, 2020-01-09:
Not sure if this is still an issue for people, but I came across this thread because I just had the same issue. I turned bluetooth of and on again and that seemed to fix it, but it seems to come back periodically. 

This only happens on my Pixel 3. Ever since Android 10 I've had tons of bluetooth issues so I find myself restarting bluetooth like this regularly. I have also been testing with a super cheap Samsung tablet that I got for free somewhere, and that has never had this problem.

RitterD, 2020-01-14:
@HansKrywaa Did you initialize bluetooth?
My code order is: ask user for location service ---> initialize bluetooth ---> start scanning
Add a button and eventlistener for every step and check the status result in the console.
what kind of device do you use? For Android 8 and latest iOS the plugin works perfectly

mdailor, 2020-10-05:
I'm having the sam problem as @ebk46 , Pixel 3a with Android 10 starts returning "Application registration failed" randomly (could be weeks or months between incidents), having the user turn bluetooth off and on fixes it. I'm just going to check for that error and throw up a message box asking the user to do just that, pretty sure anyone who uses bluetooth with their earbuds, mouse, etc. has plenty of practice with these random annoyances.