randdusing/cordova-plugin-bluetoothle #381: bluetoothle.reconnect question

Aaronwong94, 2016-11-15:
I want to use the reconnect function, where should I put the reconnect function? I had tried to put it in the bluetoothle.initialize successful callback, but it return an error "Never connected to device". I think it is because i hard code the reconnect "address" in params.
my code:
bluetoothle.reconnect(function reconnectSuccess() {
            myApp.alert("Reconnect Success!");
        }, function writeError(err) {
            myApp.alert("Error:" + err.code + "" + err.message)
        }, { "address": "E0:76:D0:37:4D:7C" }),
        setTimeout(function () {
            bluetoothle.close(closeSuccess, closeError, { "address": "E0:76:D0:37:4D:7C" })
        }, 10000);
what is the proper sequence for reconnect function? I was new in app develop. Thx for Help

estvmachine, 2016-11-16:
Always try with isConnected first, then connect, if fails reconnects.

Its just paired, then you need reconnect.

These are some use of reconnectf i have found.

Aaronwong94, 2016-11-16:
What If I want it to : 
1. scan
2.. connect to device,
3. when the bluetooth device too far from the mobile and disconnected. 
4.when I get into the range of the device and it would perform auto reconnect.
Is that possible for using the bluetoothle.reconnect? or set the bluetoothle.connect({params}] to "autoConnect" : true will do?

estvmachine, 2016-11-16:
Oh the truth i dont try with the param autoConnect, let me test and tell you.

randdusing, 2016-11-17:
I recommend just using ```connect()``` and never ```reconnect()```. If the device disconnects or there's some other type of unrecoverable error, call ```close()``` and then start the connect process over via ```connect()```. And ```autoConnect: true``` should work for what you're trying to do.

phatpaul, 2021-03-12:
I'm struggling to understand the central connection life-cycle.  I'm coming from the old and dead Evothings plugin, which was just simply connected or disconnected.  Since I switched to this plugin, I notice the connection behavior is much more advanced.

For example, If I'm connected to my BLE peripheral, but go out of range (or power-cycle my peripheral) and return, then I can no longer discover my peripheral.  From the peripheral's log output, I see that my central did actually reconnect, but my App's logic doesn't know that or initiate it.  I'm guessing I need to keep looking for the original onConnect callback?

Can you explain the difference between isConnected and wasConnected and when/why to use each?

My goal is to have auto-reconnect working as the OP described.  I also want to support paring in the future, but will need to support my legacy devices that don't support paring as well. 

I would also like to auto-connect to the last connected peripheral if my app is closed and reopened.  In that case the callbacks are no longer registered, so I'm guessing I need to store some state at the app level?

 Is there a good example or write-up of achieving this?