randdusing/cordova-plugin-bluetoothle #382: Popup request pair need discover device, how without connect first?

estvmachine, 2016-11-16:
Hi im searching how make always appear the popup of request pair bluetooth, for that i need: 

(http://stackoverflow.com/questions/26143680/bring-bluetooth-pairing-dialogue-to-the-front)
1. If the device was in discoverable mode recently
2. If the device was discovering recently
3. If the device was picked in the device picker recently
4. If Bluetooth Settings is visible

But when i use the bluetoothle.discover(), say i need to connect first. So how can i do this : discover and bond, then connect or whatever one want?

randdusing, 2016-11-17:
```startDiscovery()``` is unrelated to this plugin's ```discover()```. ```discover()``` is used to discover the BLE device's services and characteristics. If you want to the prompt the bonding popup/notification (depends on version and vendor), use ```bond()```. ```connect()``` and ```discover()``` can wait until after bonding.

estvmachine, 2016-11-17:
The problem with bond(), its internally exist a logic between show the popup for request bluetooth or show a notification. What i really need is always show the popup.  You know how i can do that?, this plugin which version of android can assure that behavior?

randdusing, 2016-11-17:
Yah, I don't think that's possible based on what I see in the API.

estvmachine, 2016-11-17:
Ok,  thanks with this plugin we cant assure that behavior.

closed 2016-11-17 by estvmachine