randdusing/cordova-plugin-bluetoothle #383: iOS still requires disconnect for proper disconnection.

kgillespieanaren, 2016-11-18:
The documentation indicates that calling the bluetoothle.close function is enough to properly disconnect this is the case on Android, but not iOS.

On iOS device I must call disconnect first and then close. Not sure if you want to look into it or simply update the documentation to reflect the case with iOS.

kgillespieanaren, 2016-11-18:
Oh tested with the latest version 4.3.0 with Phonegap 6.4.0

randdusing, 2016-11-19:
I think it's a new issue as of iOS 10. Easiest solution is to just ensure disconnect is called first. I've updated the readme...hopefully people will read it :)

closed 2016-11-19 by randdusing

kgillespieanaren, 2016-11-20:
Thanks! I think I'm just going to call disconnect no matter what from now on. =P Hopefully doesn't hurt anything to call it before a close right?