randdusing/cordova-plugin-bluetoothle #385: start scan dosen't find devices in some device(Android)

arf54, 2016-11-19:
Hello,
I created the simple cordova application project that uses Bluetoothle .But this app works in some devices and dosen't work in some devices(dosen't find bluetooth devices ).
the strange thing that when I create this project with intel XDK plat form,it  works!.
since I should use Cordova platform for main project what should I do or modify for this app?

ccravens, 2016-11-24:
I can also confirm this to be the case. The android debug statements for me are:

```
11-23 21:45:11.723 14573-14586/com.ionicframework.ses425937 W/Binder: Caught a RuntimeException from the binder stub implementation.
                                                                      java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results
                                                                          at android.os.Parcel.readException(Parcel.java:1620)
                                                                          at android.os.Parcel.readException(Parcel.java:1573)
                                                                          at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:890)
                                                                          at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onClientRegistered(BluetoothLeScanner.java:375)
                                                                          at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:56)
                                                                          at android.os.Binder.execTransact(Binder.java:453)
11-23 21:45:13.743 14573-14573/com.ionicframework.ses425937 I/chromium: [INFO:CONSOLE(7)] "Start Scan Success : {"status":"scanStarted"}", source: file:///android_asset/www/js/services.js (7)
11-23 21:45:13.753 14573-14573/com.ionicframework.ses425937 D/cr_Ime: [ImeAdapter.java:253] updateKeyboardVisibility: type [0->0], flags [0], show [true], 
11-23 21:45:13.753 14573-14573/com.ionicframework.ses425937 D/cr_Ime: [ImeAdapter.java:326] hideKeyboard
11-23 21:45:13.753 14573-14573/com.ionicframework.ses425937 D/cr_Ime: [InputMethodManagerWrapper.java:56] isActive: true
11-23 21:45:13.753 14573-14573/com.ionicframework.ses425937 D/cr_Ime: [InputMethodManagerWrapper.java:65] hideSoftInputFromWindow
11-23 21:45:13.753 14573-14573/com.ionicframework.ses425937 D/SystemWebChromeClient: file:///android_asset/www/js/services.js: Line 7 : Start Scan Error : {"error":"startScan","message":"Scanning already in progress"}
11-23 21:45:13.753 14573-14573/com.ionicframework.ses425937 I/chromium: [INFO:CONSOLE(7)] "Start Scan Error : {"error":"startScan","message":"Scanning already in progress"}", source: fil
```

randdusing, 2016-11-24:
@ccravens You need to request permissions and have the user accept, or you'll receive that error when scanning 'unpaired' devices.

@arf54 What devices does it work and not work on?

coderofsalvation, 2016-11-24:
I had the same problem on android6 but it worked when I turned on the global location-switch (otherwise allowing location for the app won't have any effect)

arf54, 2016-12-03:
@coderofsalvation ,Thank you, that was my problem! Location switch in Permissions app setting was off!
after switch it on ,it works.;-)