randdusing/cordova-plugin-bluetoothle #386: Getting the address of an already paired device

soptrau, 2016-11-29:
Hello,
I'm writing an app which utilizes an external input device, which is paired with my smartphone via bluetooth. Since I don't want to reinvent the wheel, I figured that I could use the platforms native pairing logic to connect my smartphone with my input device. So, the devoce is already paired with the input device once my app starts.
From what I gather from your documentation, every device related opperation (reading/writing messages, dicovering services etc) requieres a specific device address as a parameter to work, which I only seem to obtain from scanning for bluetooth devices (which I rather want to avoid since the devices are already paired). This would also require me to properly identify my device, which requires additional expenditure.

Is there a quicker and easier way to obtain the address of an already bonded device?

randdusing, 2016-11-29:
Try using ```retrieveConnected()```

soptrau, 2016-11-29:
Thanks for your fast reply.

> Try using retrieveConnected()

retrieveConnected() returns an empty array, since I don't know the services, therefore my params-object is empty. To discover those, I need the address (if I get your Documentation right). Correct me if I'm wrong, but right now this seems to be kind of a chicken-egg problem.
I'm currently testing on a Samsung Galaxy S6 running Android 6.0 and I'm pretty new to the whole Bluetooth thematic in general, so maybe I'm missing something major.
Are those Service ID's regulated by any international standart? If so, I could consult the devices tech-listings, maybe I'll find something useful there.

randdusing, 2016-11-29:
You don't need to specify the services on Android. Are you sure they are bonded? Internally the Android plugin calls getBondedDevices() https://developer.android.com/reference/android/bluetooth/BluetoothAdapter.html#getBondedDevices()

Are your devices BT Classic or BLE?

soptrau, 2016-11-29:
>Are you sure they are bonded?
Yeah, this device features an alternative mode, to directly write the scanned information as a "bluetooth keyboard" into textfields, which is working fine.

>Are your devices BT Classic or BLE?
Both are BLE

randdusing, 2016-12-01:
Hmm, have a link to the device's website? Maybe I can take a look

soptrau, 2016-12-01:
http://www.panmobil.com/produkte/smartscanndy/

That's the link to the device's homepage.
Right now, I am building a workaround where I connect with the device "again", so that I could discover the services. So far it works, but it's far from optimal, therefore please keep me updated if you find a better solution. 
Again, I am pretty new to this whole matter, so I might be missing something obvious but thank you for your help :)

randdusing, 2016-12-01:
I don't see any mentions of Bluetooth LE, Smart or 4.0

soptrau, 2016-12-01:
> **Overview of the most important features**
> ...
> Bluetooth (SPP und HID) (optional)

We bought the variant with included Bluetooth functionallity. But that might actually be the problem, since I don't find the "LE"-part mentioned anywhere.

soptrau, 2017-01-05:
I just started back from scratch and intrestingly enough didn't encounter the same problem, since I tried a different approach.

Thanks again for your help.

closed 2017-01-05 by soptrau