randdusing/cordova-plugin-bluetoothle #387: iOS write does not work

JacobYoungDigi, 2016-11-30:
Everything works perfectly on Android, but when I try to run the same code on iOS, my program gets hung up on the write. Here is my code:

`bluetoothle.initialize(initializeSuccess, initializeError, {
        "request": true,
        "statusReceiver": false,
        "restoreKey": "bluetoothleplugin"
});

function initializeSuccess(obj) {
	if (obj.status == "enabled") {
		console.log("Bluetooth initialized successfully.");
	}
	else {
		console.log("Bluetooth disabled.");
	}
};

function initializeError(obj) {
	console.log("Bluetooth initialize error.");
};

var address = *valid MAC address*
bluetoothle.connect(connectSuccess, connectError, { "address": address });

function connectSuccess(obj) {
	if (obj.status == "connected") {
        console.log("Connected");
	}
	else if (obj.status == "connecting") {
        console.log("Connecting");
	}
	else if (obj.status == "isconnected") {
		console.log("Device already connected");
	}
	else {
		console.log("Device disconnected");
	}
}

function connectError(obj) {
	console.log("Connection error");
}

bluetoothle.discover(discoverSuccess, discoverError, { "address": address });

function discoverSuccess(obj) {
	console.log("Discovery success: " + JSON.stringify(obj));
}

function discoverError(obj) {
	console.log("Device not connected");
}

var request = "Hello";
var encodedString = bluetoothle.bytesToEncodedString(request);

var params = {
	"address": address,
	"service": *valid service ID*,
	"characteristic": *valid request ID*,
	"value": encodedString
};
		
bluetoothle.write(function (obj) {
		console.log("Write success: " + bluetoothle.encodedStringToBytes(obj.value));
	}, function (obj) {
		console.log("Write error: " + obj.message);
	}, params);`

I can confirm that I get a successful initialize message, a successful scan message, a successful connect message, and a successful discovery message with all the services and characteristics displayed, The only problem is that neither of the callback functions in the bluetoothle.write are called. Suggestions?

JacobYoungDigi, 2016-11-30:
Never mind. I just had to add "type": "noResponse" to the params

JacobYoungDigi, 2016-11-30:
Fixed. In my actual code, my write function either resolves or rejects a promise in the callbacks. So to fix this I just resolved the Promise after the call to write. Kind of a hacky way to do it, so if you have a suggestion on how to handle errors and successes with iOS that would be appreciated.

This is my _actual_ code:
return new Promise(function (resolve, reject) {
            bluetoothle.write(function (obj) {
                console.log("Write success: " + bluetoothle.encodedStringToBytes(obj.value));
                //myApp.alert("Write success");
                //addLogMsgToScreen("Write success: " + bluetoothle.encodedStringToBytes(obj.value));
                resolve();
            }, function (obj) {
                console.log("Write error: " + obj.message);
                //myApp.alert("Write error: " + obj.message);
                //addLogMsgToScreen("Write error: " + obj.message);
                reject();
                myApp.alert("Check that your RCI request is correct", "<b>Write Failed</b>", function () {
                    mainView.router.back();
                });
            }, params);
            resolve(); // this is the line of code
        });

randdusing, 2016-12-01:
Yah that kind of defeats the purpose of using a promise :)