randdusing/cordova-plugin-bluetoothle #388: iOS occasional write / subscribe error

CaptainJeff, 2016-12-07:
I am experiencing an issue on iOS where 1 out of every 15 or so sync's to my device fails. The device does connect and subscribe and executes the first 4 writes and then I receive two errors. The first one is a Bluetooth write failure, where the error message is "Device is disconnected" immediately followed by a subscribe error with the same error.

So it seems as though it is disconnecting at some point during this process. It isn't recoverable. The next 10 or so attempts to reconnect fail and then all of a sudden it'll work.

Is there any error you are aware of that would cause the device to disconnect during this process? The responses to the write requests seem to be successful and then all of a sudden it just disconnects.

randdusing, 2016-12-08:
Seems like an issue with the device. Do you have another mobile device you could try connecting to the BLE device with? Or can you control the BLE device itself?

CaptainJeff, 2016-12-08:
I tried on iphone 6 and iPhone 5 and had the issue on both. I couldn't replicate the issue on android though

randdusing, 2016-12-09:
When an iOS device is causing problems, see if your Android can connect to it.

CaptainJeff, 2016-12-09:
Yeah android can connect fine. Its just happening on iOS

randdusing, 2016-12-10:
I mean try to connect simultaneously. If the Android can't connect after the iOS device lost connection, perhaps it's something wrong with the device (that's only triggered when connected to iOS)

CaptainJeff, 2016-12-14:
It seems like the android can connect after the iOS device loses its connection. We are receiving an error whenever this happens that I hadn't noticed prior. It says:

Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly." UserInfo={NSLocalizedDescription=The connection has timed out unexpectedly.}

Any thoughts on what would be causing that? Also, let me know if there is anything you'd like me to test

CaptainJeff, 2016-12-14:
Correction: There are two errors. The one I mentioned above Code 6 and also Code 0. There doesn't seem to be any consistency to when we see code 0 or code 6

ERROR: Bluetooth Service. Subscribe Error : {"error":"subscription","message":"Error Domain=CBErrorDomain Code=0 \"Unknown error.\" UserInfo={NSLocalizedDescription=Unknown error.}","characteristic":"FFF7","name":"test1","service":"FFF0","address":"9BCD0987-0D52-53BD-2186-1DC86D3B2412"}

randdusing, 2016-12-16:
Do you control the BLE device? The post below leads to a timing issue on the iOS device that sounds like it could be fixed if your BLE device can send different timing parameters. But I'm not sure if this is the only reason for the disconnect. The code 0 zero is really hard to say since it's an unknown error. 
http://stackoverflow.com/questions/32588325/ios-bluetooth-le-code-6-the-connection-has-timed-out-unexpectedly

You could try logging at a lower level and see if you find anything interesting. A bit annoying to set up, but I want to say the iOS legs are pretty helpful. https://developer.apple.com/bug-reporting/profiles-and-logs/

Prathapreddy26, 2019-03-18:
Anyone facing this issue, Error Domain=CBATTErrorDomain Code=6 "The request is not supported." UserInfo={NSLocalizedDescription=The request is not supported.} I am doing research on this error from past 4 days. I didn't get any solution. Can anyone please help me out for this?