randdusing/cordova-plugin-bluetoothle #395: Detect "Busy"

unclespode, 2017-01-03:
If I have multiple read/writes to do, I use my own queuing system to handle this. However, with the async nature of things - this isn't always possible. If I try and read/write when another operation is happening it will error out. 

Is there a way of checking if the device is free/busy before I make any calls?

randdusing, 2017-01-04:
No current way. Android and iOS both queue now at the plugin level, does that not work for you?

unclespode, 2017-01-04:
I did see that in the documentation, but it wasn't described as if I needed to do something manually or it was automatic. I'll have to try that.

The issue I'm having is trying to subscribe at the same time as reading - this seems to cause an error.