randdusing/cordova-plugin-bluetoothle #396: Mobile app doesn't detect every iBeacon advertisment

MarinRadmilovic, 2017-01-17:
Hey! This plugin is great and very easy to use. Congrats on that. I made a simple application that detects iBeacon when near and everything works fine except one thing. I use startScanWithOptions and reportDuplicates is set to true. I set beacon advertising frequency on 500ms. The problem is that startScanWithOptions callback is called every 5-6 seconds for one beacon, and not every 500ms. Is it problem with Bluetooth hardware or with this plugin, and is there any way to get every message. Thank you very much and keep up the good work.

randdusing, 2017-01-18:
Using iOS? Is your app in the foreground or background?

MarinRadmilovic, 2017-01-18:
Thanks for fast reply. I'm using Android 6.0.1. and app is working in the foreground. I had the same problem before but not with mobile or tablet but with BeagleBone. The same wireless chip was used for wifi and bluetooth communication so it was multiplexing messages. Maybe it's something like that but I dont know.

randdusing, 2017-01-18:
allowDuplicates doesn't apply to Android. Instead you need to use the Android-specific variables. You probably want to set the scan mode to low latency

MarinRadmilovic, 2017-01-18:
I think I did :). I'm using ng-cordova and your angular wrapper. 

`	$cordovaBluetoothLE.startScan({"services":[], "allowDuplicates": true, "scanMode": 2}).then(null, ...)
`
I think this line of code set's it to low latency. But what I'm trying to do is to know when my mobile enters beacon range, and when it gets out of it. So I created a timers that are renewed each time application recieves Beacon advertisement. When beacon is in range his address is in nearbyDevices array. If application doesn't recieve advertisement from beacon in 15 seconds application removes it from nearbyDevices. That's the core. 
Problem is that sometimes my beacon is in range but application recieves advartisement every 20 seconds, and not every advertisement (beacon advertising frequency is set to 500 ms). But I wrote all that before. I just ask if there is some other way to know if mobile is in beacon range or not because with this kind of "random" behevior this will not work. 

P.S. I cant connect to beacon beacuse it is possible to be in range of multiple beacons and I want to know that.

Thank you for you help.

randdusing, 2017-01-19:
Actually all the default settings are pretty aggressive. I need to update the readme to be more accurate. Do you have any other BLE devices you can try this with? Perhaps it's something weird with the devices themselves. Or try on an iOS device?

MarinRadmilovic, 2017-01-24:
Sry. It was my bad. I was using Google Nexus 7 and I didn't check specifications. It is using Bluetooth 3.0 and not 4.0. I tried on another device and it is working perfectly. Thank you for help.

closed 2017-01-24 by MarinRadmilovic