randdusing/cordova-plugin-bluetoothle #397: Can't see bluetooth barcode scanner

dpeeters, 2017-01-20:
I am working on a app that connects with a RIDA barcode scanner and I am testing on an Acer Iconia One tablet with Android 6.0. 
The app can't find the RIDA scanner with the bluetoothle plugin, but I am able to see other devices.
I turned on the global location-switch but no luck. Is there anything else I should do that might help?

Scanner spec: http://www.datalogic.com/upload/marketlit/datasheets/DS-RIDA-DBT6400-EN.pdf

randdusing, 2017-01-20:
Does it show up in Android's native Bluetooth settings?

dpeeters, 2017-01-20:
Yes, I can also pair with the scanner. It doesn't show up in the foundDevices.
Did you come across a barcode scanner that works well with this plugin?

randdusing, 2017-01-21:
Try retrieveConnected in addition to startScan. And can you confirm it's advertising as a BLE device? Maybe it's using Bluetooth Classic

maximumgs, 2017-02-06:
This sounds similar to my issue. I am using this plugin to work with an hands-free earpiece on iOS. I am able to see the device within the Bluetooth Settings, but it does not show up in the found devices. I have checked the specs for the earpiece and it is indeed BT LE 4.1.

@randdusing You mentioned above "advertising as a BLE device". Is it possible and/or common for BT LE 4.x devices to "advertise" using Bluetooth Classic as opposed to the newer standard?

Thanks in advance for any help.

randdusing, 2017-02-06:
I believe it is possible. Can you use an app like LightBlue Explorer to see if exists?

maximumgs, 2017-02-06:
Yeah, I tried that already and it does not see the device. It sees my Fitbit, as does your library, but not the device. Neither your library nor LightBlue see my Bose Soundlink Headphones or my Jawbone Big Jambox either. I  pretty sure the Jambox is BT 2.x. 

Your library and LightBlue seem to find the same device and are consistent. Would you expect to find my other devices?

Again, thanks for your help and your quick response.

randdusing, 2017-02-07:
Just noticed you said earpiece. I'm willing to bet it's not BLE since BLE doesn't currently have enough throughput for audio.

maximumgs, 2017-02-07:
You are correct. I spent a big chunk of time today digging into all that is Bluetooth. The earpiece is Bluetooth Classic and supports only standard profiles (HFP, A2DP and AVRCP). Unfortunately it looks as though there are no Cordova plugins capable of communicating with these profiles. Even natively in iOS this looks like a less than reasonable endeavor.

Once again, thanks for your help and responsiveness.

dpeeters, 2018-09-20:
@amalshah71 Yes, DBT6400 works great.

amalshah71, 2018-09-24:
@dpeeters :: Thanks for the reply, But i wonder how did it work...Since for me even LightBlue app is not able to detect DBT6400 neither on Android nor iOS...Which OS you could get it working on ?