randdusing/cordova-plugin-bluetoothle #403: Plugin not found for OSX target

mo-seph, 2017-02-10:
I can't get the plugin to deploy on OSX. When I run on my android device, it works, and can check the Bluetooth enabled status. Running on my laptop (OSX 10.10.5) I get the message "HelloWorld[12629]: ERROR: Plugin 'BluetoothLePlugin' not found, or is not a CDVPlugin. Check your plugin mapping in config.xml."

I'm running Cordova 6.5.0. I've tried the version from the Cordova site, and now I'm trying the version straight from the repo.

config.xml is:
```<?xml version='1.0' encoding='utf-8'?>
<widget id="com.example.hello" version="1.0.0" xmlns="http://www.w3.org/ns/widgets" xmlns:cdv="http://cordova.apache.org/ns/1.0">
    <name>HelloWorld</name>
    <description>
        A sample Apache Cordova application that responds to the deviceready event.
    </description>
    <author email="dev@cordova.apache.org" href="http://cordova.io">
        Apache Cordova Team
    </author>
    <content src="index.html" />
    <plugin name="cordova-plugin-whitelist" spec="1" />
    <access origin="*" />
    <allow-intent href="http://*/*" />
    <allow-intent href="https://*/*" />
    <allow-intent href="tel:*" />
    <allow-intent href="sms:*" />
    <allow-intent href="mailto:*" />
    <allow-intent href="geo:*" />
    <platform name="android">
        <allow-intent href="market:*" />
    </platform>
    <platform name="ios">
        <allow-intent href="itms:*" />
        <allow-intent href="itms-apps:*" />
    </platform>
    <engine name="osx" spec="~4.0.1" />
    <engine name="android" spec="~6.1.2" />
    <plugin name="cordova-plugin-bluetoothle" spec="https://github.com/randdusing/cordova-plugin-bluetoothle.git" />
</widget>```

randdusing, 2017-02-10:
Are you installing fromt he command line or by editing config.xml?

mo-seph, 2017-02-10:
Command line - 'cordova plugin add https://github.com/randdusing/cordova-plugin-bluetoothle.git --save'

randdusing, 2017-02-10:
Does it happen every time on startup? I remember running into this issue previously but it was sporadic and I believe it affected all plugins. Not sure if it's an issue with OSX/Cordova or the plugin itself.

mo-seph, 2017-02-10:
Every time. I'll try working with some different plugins, though, and check whether it's my setup.

mo-seph, 2017-02-10:
It seems to be all plugins. Just to check - does this plugin work with OSX?

randdusing, 2017-02-11:
It has definitely worked before, but randomly all plugins would fail. It has been a while since I last tested. I was hoping it was a Cordova OSX issue.

mo-seph, 2017-02-11:
I think it is - the File plugin doesn't work, so it looks like a setup issue. I'll report back if I get things going

mo-seph, 2017-04-12:
Just to follow up on this - it seems to be a general cordova/osx issue. I've reported it here: https://issues.apache.org/jira/browse/CB-12448

My current workaround is to remove and re-add the osx platform with every build.

closed 2017-04-12 by mo-seph