randdusing/cordova-plugin-bluetoothle #404: subscribe massive data

pineson, 2017-02-14:
My app need to receive 10k data by notification, I try in different androids, after 3k, some data will be lost, any suggestion on how to receive the whole data safely? Thanks.

randdusing, 2017-02-14:
Check out the file transfer example in the Angular wrapper. Subscribe isn't completely reliable, so you'll need to build your own reliable protocol on top of it.

pineson, 2017-02-15:
I wonder if the subscribe cache is 4096 bytes, after receiving 4k bytes, the speed of subscribe result is slow down obviously, and this is the time it begins to miss some bytes, is there a way to enlarge the cache or clear the cache to make it work faster?

randdusing, 2017-02-15:
Are you just blasting out 4KB? The receiver may not be able to handle it. Instead blast out a few chunks, wait for an acknowledgement from the receiver and then repeat.

pineson, 2017-02-16:
I am talking about the receiver, now it works like subscribe, receive 1k, save to localstorage, then write a signal to sender for next 1k. repeatedly, usually, after receive 4k, some data will be missed, can't get the whole 1k and log is obviously slower than before, so any idea that can secure the 1k received?

randdusing, 2017-02-17:
Not really following. BLE isn't a reliable protocol, so you'll have to build reliability on top of it.