randdusing/cordova-plugin-bluetoothle #405: Error Domain Code=10 on subscribe (iOS 10.2)

OnkarBhattacharya, 2017-03-01:
Hello @randdusing ,

So, we have a in-house developed device with BLE chip. At this point, I am working on the development of an iOS app for interfacing with the device.
The issue I am encountering sounds similar to #373.
I will try to be as clear as possible :

1. The iOS app is able to successfully scan for surrounding BLE devices  
2. It can connect to the selected device and get its services and characteristics (using services+characteristics and discover)
3. After connect + discover, it is able to read characteristics with {read : true} and write to characteristics with {write with response : true & write without response : true}
4. However, i am unable to execute a subscribe of a characteristics with {notify : true}. The error message is 
```
> error: "subscription"
message: "Error Domain=CBATTErrorDomain Code=10 \"The attribute could not be found.\" UserInfo={NSLocalizedDescription=The attribute could not be found.
```
The hardware has fixed characteristics to write commands to and get response.
eg : 
service UUID : 6E400001-XXXX-XXXX-XXXX-XXXXXXXXXXX
characteristic UUID _(02)_ : 6E400002-XXXX-XXXX-XXXX-XXXXXXXXXXX : write without response : true
characteristic UUID _(03)_: 6E400003-XXXX-XXXX-XXXX-XXXXXXXXXXX : notify : true
so to get temperature value, a command needs to be sent to characteristic with _"02"_ and then it the device gives a response on characteristics _"03"_

The flow implemented in my code can be briefly summarised so : 
- scan for peripherals -> successful
- on select of peripheral, connect to it (saving proper reference to the device) -> successful
- discover services / characteristics / descriptors -> successful
- send command to _"02"_  -> successful
- subscribe to _"03"_ -> error

Debugged everything possible on the hardware end, tried communication via LightBlue , everything looks good. However on iOS, no success.

Any help will be appreciated!

P.S : thanks for the plugin

randdusing, 2017-03-03:
Can you post the JSON output from the discover call? And are you calling both services+characteristics and discover? That could be causing issues.

OnkarBhattacharya, 2017-03-06:
> Object
> 
> address: "91A830E7-8A1D-4B7A-BEBD-5DD655D41A56"
> 
> name: "WIKA CPG1500"
> 
> services: Array (2)
> 0 Object
> 
> characteristics: [{uuid: "2A29", properties: {read: true}, descriptors: []}, {uuid: "2A24", properties: {read: true}, descriptors: []}, {uuid: "2A25", properties: {read: true}, descriptors: []}] (3)
> 
> uuid: "180A"
> 
> Object Prototype
> 1 Object
> 
> characteristics: Array (4)
> 0 {uuid: "6E400002-B5A3-F393-E0A9-E50E24DCCA9E", properties: {writeWithoutResponse: true}, descriptors: []}
> 1 {uuid: "6E400003-B5A3-F393-E0A9-E50E24DCCA9E", properties: {notify: true}, descriptors: []}
> 2 {uuid: "6E400004-B5A3-F393-E0A9-E50E24DCCA9E", properties: {write: true}, descriptors: []}
> 3 {uuid: "6E400005-B5A3-F393-E0A9-E50E24DCCA9E", properties: {read: true}, descriptors: []}
> 
> Array Prototype
> 
> uuid: "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"
> 
> Object Prototype
> 
> Array Prototype
> 
> status: "discovered"
> 
> Object Prototype

randdusing, 2017-03-07:
There should be the client characteristic configuration descriptor for the characteristics with notification enabled.

OnkarBhattacharya, 2017-03-07:
Thank you for the help. I will try this option out and close the issue if solved.

closed 2017-03-08 by OnkarBhattacharya

weihuaronaldo, 2018-03-02:
Hi @OnkarBhattacharya, have you ever been solved this problem? If so, how to solve it ?

OnkarBhattacharya, 2018-03-02:
@weihuaronaldo Yes we did manage solving this problem. The changes to be implemented were on the hardware side and the solution is "There should be the client characteristic configuration descriptor for the characteristics with notification enabled".  The app did not need any changes as this plugin works perfect after Hardware service+characteristics adjustments.

weihuaronaldo, 2018-03-05:
Thanks! In our case, the problem comes when we use the newest 5th generation ipad with LightBlue, while other devices (such as iphone6s, ipad mini 2) is OK. Any ideas about this? Is there any changes about the BLE inside ios OS ?

jaydeepc, 2021-09-14:
@OnkarBhattacharya a little late in this thread but facing the same issue. Do you remember what exact characteristics you were setting? We are trying to connect iphone 12 to an esp32 and from esp32 below is what we are setting:
`
void startBLE()
{
  Serial.println("Starting BLE work!");

  BLEDevice::init("Rolling Pawn");
  pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  pCharacteristic = pService->createCharacteristic(
      CHARACTERISTIC_UUID,
         BLECharacteristic::PROPERTY_READ |
          BLECharacteristic::PROPERTY_WRITE |
          BLECharacteristic::PROPERTY_NOTIFY);

  pCharacteristic->setValue("Hello World says Neil");
  String chipId = String((uint32_t)(ESP.getEfuseMac() >> 24), HEX);
  pCharacteristic->setValue(chipId.c_str());
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06); // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}
`

peitschie, 2024-01-29:
Just reporting back, the specific addition required for an Arduino based system using the older Bluetooth stack is adding this descriptor to the characteristic:
```
pCharacteristic->addDescriptor(new BLE2902());
```

This is based of these examples from the Arduino library itself ([here](https://github.com/espressif/arduino-esp32/blob/cceebb58f170c34a2356f0d22106dd6f0d141023/libraries/BLE/examples/Notify/Notify.ino#L76)) and the [BLE C++ Guid.pdf](https://github.com/nkolban/esp32-snippets/blob/master/Documentation/BLE%20C%2B%2B%20Guide.pdf).